mavlink/qgroundcontrol #7878: Complete UI window missing with latest daily

MaEtUgR, 2019-10-06:
## Expected Behavior
The QGC window opens when the `.exe` on Windows or the `.AppImage` on Ubuntu is started.

## Current Behavior
No window shows up when starting QGC. The application seems to be running since I:
- see the Application running as background application in Windows's task manager
- see the established MAVLink data link on the vehicle
- hear the audio output of QGC announcing the flight mode
- see the error message when trying to open another instance without killing the original one

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Load latest daily from https://docs.qgroundcontrol.com/en/releases/daily_builds.html for Windows or Linux
2. Start
3. Wait for the window to (not) appear

## System Information
- Operating System: Windows 10 1803, also verified on Ubuntu 18.04.1 VM
- QGC Version: The latest daily downloaded when posting this issue. I can't see the version since it doesn't start and there's neither a `--version` parameter nor a splash output at the beginning of stdout when starting.
- Flight Controller: SITL
- Autopilot (with version): PX4 latest master `941a3258b6e87640`

## Log Files and Screenshots
Here's the console output on Linux:
```
maetugr@ubuntuvm:~$ ./QGroundControl.AppImage.1
Settings location "/home/maetugr/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\n"
GStreamer debug output: "/home/maetugr/Documents/QGroundControl/Logs/gst/gstreamer-log.txt"
Error loading text-to-speech plug-in "speechd"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
Failed to set Geoclue positioning requirements. Geoclue error: org.qtproject.QtDBus.Error.InvalidObjectPath
QGCPositionManager error 2
MAVLinkLogManagerLog: MAVLink logs directory: "/home/maetugr/Documents/QGroundControl/Logs"
System reported locale: QLocale(English, Latin, UnitedStates) "en_US"
Loading localization for "en_US"
Qt localization for "en_US" is not present
Error loading application localization for "en_US"
Map Cache in: "/home/maetugr/.cache/QGCMapCache300" / "qgcMapCache.db"
QQmlApplicationEngine failed to load component
qrc:/qml/MainRootWindow.qml:275 Type LogReplayStatusBar unavailable
qrc:/qml/QGroundControl/Controls/LogReplayStatusBar.qml:59 Type QGCComboBox unavailable
qrc:/qml/QGroundControl/Controls/QGCComboBox.qml:12 module "QtQuick.Controls" version 2.11 is not installed

Adding target QHostAddress("127.0.0.1") 14570
Switching outbound to mavlink 2.0 due to incoming mavlink 2.0 packet: 0x1b1b288 1 2
Temp log "/tmp/FlightData146713.mavlink"
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters cache match failed /home/maetugr/.config/QGroundControl.org/ParamCache/1_1.v2
_parametersReady true
```

dogmaphobic, 2019-10-07:
This is the reason:

`qrc:/qml/QGroundControl/Controls/QGCComboBox.qml:12 module "QtQuick.Controls" version 2.11 is not installed`

That Qt plugin is missing.

DonLakeFlyer, 2019-10-07:
Ah, crap I was building with Qt 5.12 where it works. And then CI tests don't run ui so that all works fine.

MaEtUgR, 2019-10-08:
Tested on Windows. UI shows again with latest
[image]
Thanks for fixing! 👍

closed 2019-10-08 by MaEtUgR