mavlink/qgroundcontrol #7897: Andriod and Ardupilot VTOL Takeoff and Transition

cody303, 2019-10-09:
It seems as though using Andriod QGroundControl and ardupilot the VTOL Takeoff and Transition waypoint uploaded by QGroundControl is being ignored by the Ardupilot Firmware. This causes the plane to transition to fixed wing on the ground.

DonLakeFlyer, 2019-10-15:
If ArduPilot doesn't fail the mission upload then that would be a firmware problem. Need to talk with ArduPlane folks. Possible that those commands have bad values in them though. Attach a plan file and I can take a look.

DonLakeFlyer, 2019-10-16:
So there are some problems here:
* VTOL_TAKEOFF for ArduPlane does not take a lat/lon
* VTOL_TAKEOFF for ArduPlane does not take a heading

But from looking at the code in ArduPlane it just ignores those so it shouldn't be a problem. I will fix though.

The attached Plan file has a good relative altitude in it of 50m so that should work fine. As far as I can tell this is a possible firmware problem. @tridge Anything special need to happen with the VTOL_TAKEOFF command on a mission?

[Antone.LUPINE.plan.txt](https://github.com/mavlink/qgroundcontrol/files/3735598/Antone.LUPINE.plan.txt)

DonLakeFlyer, 2019-12-11:
I fixed VTOL_TAKEOFF to not take a location which is more correct. But like I stated above this shouldn't have affecting anything as far as I can tell in the firmware code. So from what I can see this isn't a QGC problem.

PR #8104 (2019-12-11): Random fixes
Fix #7897, #8078, #7887
closed 2019-12-11 by DonLakeFlyer