mavlink/qgroundcontrol #7899: New Plan Wizard(s): can't get the wizards back after "X"ing out

Antiheavy, 2019-10-10:
After hitting the "X" in the upper right corner, you can't get the wizards back unless you exit the application and come back in.  I think if you exit Plan screen and come back to the Plan screen this should re-appear (unless there is an existing plan loaded).
[image]

Antiheavy, 2019-10-10:
Or maybe any time you select New the "Create Plan" box should show up, even if there is an existing mission onscreen (e.g. there is an old mission on vehicle that was downloaded upon connection):
[image]

Antiheavy, 2019-10-11:
Actually, if the "New..." button wasn't grayed out, I think this would fix the problem.

DonLakeFlyer, 2019-10-15:
> Actually, if the "New..." button wasn't grayed out, I think this would fix the problem.

Yes, this is a great idea.

PR #7918 (2019-10-16): Plan: Create Plan overlay and friends
* Fix #7907 - Show correct terrain or incomplete data warnings
* Fix #7899 - New... is always enabled. Clicking New will bring up Create Plan overlay
closed 2019-10-16 by DonLakeFlyer