mavlink/qgroundcontrol #7901: Request: Plan Wizard: Takeoff default distance and rotation

Antiheavy, 2019-10-10:
in the new Plan Wizard, if a Takeoff is part of the plan, it should:

- default to be above of the Planned Home (to the North).  Just looks more correct and easier to read.
- default to a distance calculated by the vehicle's minimum ground speed, climb rate, and default Takeoff waypoint altitude. (e.g. for MC and VTOL this distance may be essentially zero because min ground speed is zero - however for fixed wing this will be an appropriate distance away from home).
[image]

Antiheavy, 2019-10-16:
Or, a simpler approach is to default the Takeoff distance to a reasonable value for most small fixed wing drones.  I suggest the default distance be 1.5X the default altitude (also, straight north looks better as a default).

tilaktilak, 2019-10-29:
I think that's needs to be FixedWing only thing. And we probably want the climb to be achievable by the particular drone, not sure how to do that ? Not sure 1.5 will work for every drones, what happen if it fails ? Stall ? Overshoot ?

Antiheavy, 2019-10-29:
> I think that's needs to be FixedWing only thing.

Yes, definitely only a fixed wing thing.  QGC has the ability to know if it is connected to a fixed wing or not (and remember most recent).

> And we probably want the climb to be achievable by the particular drone, not sure how to do that ? Not sure 1.5 will work for every drones, what happen if it fails ? Stall ? Overshoot ?

This was the reasoning behind my initial suggestion to read the autopilot parameters for things like minimum ground speed, climb rate/pitch angle, etc.  After a short discussion with @DonLakeFlyer that seemed more complicated than necessary.  My suggestion now is to choose a "reasonable" default (the current default isn't reasonable for most fixed wings), but let the user drag wherever they want after that.  Maybe in the future the default could be set in QGC's custom vehicle plugin for integrators who want a different default.

tilaktilak, 2019-10-29:
That looks great then ! :-)

DonLakeFlyer, 2019-12-11:
The takeoff distance is now based on a glide slope default which I can't quite remember. So I think this is all handled now.

closed 2019-12-11 by DonLakeFlyer