mavlink/qgroundcontrol #7902: Request: new Polygon Tools looks jarring

Antiheavy, 2019-10-10:
weird looking menu with the title on the right side and lots of empty space on the right side.
[image]

Maybe a centered version with less empty space?
[image]

tilaktilak, 2019-10-29:
I am also concerned about this new toolbar. 
First of all it does not follow the theme colors, it could be confusing to find that now the left-hand toolbar is not the main one but this one is. 

Maybe this Basic/Circular/Polygon/Load KML/SHP can replace the left-hand toolbar during the time the polygon is defined ? And can then use a similar style (black and white icon based vertical toolbar).

DonLakeFlyer, 2019-11-11:
Fixed

closed 2019-11-11 by DonLakeFlyer