mavlink/qgroundcontrol #7903: Plan Wizard: simplify left hand toolbar

Antiheavy, 2019-10-10:
Two parts:

1)
When using a Plan Wizard, the left hand toolbar could be simplified to eliminate waypoint tools and encourage people to follow the guided wizard on the right hand side of the screen.

In other words: only show left hand toolbar way point types when using a Custom Plan or when loading a saved mission:

[image]

2)
When building a custom mission (or after loading a saved mission) show additional dedicated buttons on the left had toolbar for Takeoff and Landing Pattern.  These are almost always used and should be shown to the user at the first level (not hidden):
[image]

tilaktilak, 2019-10-22:
I think there is more of a design discussion to have regarding this wizard and what is intended to do with that. 
There is two approaches that are opposed to each other here : 
- We either show a really simple and constrained way to generate a complete flight plan (with Home / transitions WP / CoverZone / ... / Landing ) 
- We have all the tools to create each of those steps of a mission flight plan.

If we still want to switch from one "approach" to another it's probably because we are trying to extra-simplify things while covering all the cases.

The only way to achieve that is by adding a lot of extra pop-up / toolbar and continue handling cases where we getting in and out of this wizard's state machine.

 This also brings a lot of corner cases and non-obvious situations ( When I pick the Waypoint tool to add a waypoint while tracing a polygon survey for instance, or trying to get out of "tracing mode" to put another waypoint after having selected another mission item on the right hand list...).

In that case we have a big popup that shows up on startup, then some of the wizard appears on the right in the mission item list, some of it appears as a new toolbar to select Polygon/Circle/... (Btw, why do we need "circle shape" or "Basic", shouldn't we use only polygon if we want to make it simple ?) then the wizard disappear. 

I really think that the wizard approach is great and I understand the purpose of it. If the decision is made we need to stick to this approach and define it with a granularity for which each type of user will find its place ( Not to constrained, not to complicated ) then get rid of the other "manual" approach or have it as a fallback that will completely disable all "wizard" features.

Antiheavy, 2019-10-24:
Thanks for you ideas/discussion @tilaktilak 

I'm not sure I agree the two approaches are opposed to each other.  I just see them as different paths that can be taken from the Create Plan pop-up menu.  Here is where my mind is at right now:

Step 1:  Create Plan shown to user if there is currently no flight plan or user selects "New".  As a suggestion: I've shown a couple more create plan options and also showed a simplified left-hand toolbar at this stage:
[image]

Step 2a:  If an option is selected that has a wizard, then start the wizard and continue to keep the simplified left-hand toolbar:
[image]

Step 2b: If an option is selected that doesn't have a wizard, e.g. Custom, or Load mission from file, then show a fully featured toolbar appropriate to the vehicle, including Takeoff and Landing:
[image]

Step 3:  after all of the planning wizard items are completed, maybe then show the full left-hand toolbar again?  I guess this would need more thought.

> brings a lot of corner cases and non-obvious situations ( When I pick the Waypoint tool to add a waypoint while tracing a polygon survey for instance, or trying to get out of "tracing mode" to put another waypoint after having selected another mission item on the right hand list...).

My thought is by now showing waypoint tools on the left when a planning Wizard is being used, then some of this can be avoided.  However, your point is a good one in that there is a weird feel to some of the planning wizard behavior currently and I think it is rooted in what you describe.

> (Btw, why do we need "circle shape" or "Basic", shouldn't we use only polygon if we want to make it simple ?)

I'll tell you that many of our agricultural customers really love the circle feature because of landscapes that look like this:
[image]

>  then get rid of the other "manual" approach or have it as a fallback that will completely disable all "wizard" features.

I'm not sure about totally getting rid of it, but maybe only make it available when the user selected the "custom" option or "load mission from file".  I guess this is basically the point of my original post above.

tilaktilak, 2019-10-24:
> I'll tell you that many of our agricultural customers really love the circle feature because of landscapes that look like this:

Haha that is fair enough ! :-) Still think that we can remove Basic though.

I think it's important to minimize the number of new popup/toolbar added by the wizard and avoid duplicate buttons with same functions. 
For instance : 
- It your proposition can the 'File' button from the toolbar be simplified as the "import flight plan" is on the main popup ?  
- Can't we simplify "Create Plan" to 1 level by adding "Polygon Survey", "Circle Survey"  as it occupies all the screen and appears to be half empty ?
- Can't "Open Saved Mission" and Import from KML be merged to an "Import" button ? 
[image]

From the other image image, what do you mean by "Use the insert function on the right hand list ? 
You mean this submenu ? 
[image]
 
Because this is not practical at all as it adds a MissionItem between 2 items in the list and appears on the map at the position of the last item (overlay it) so that you need to unfold the flight path to change the position. ( That brings removing it into question btw ).

Antiheavy, 2019-10-28:
> Still think that we can remove Basic though.

I don't like the term "basic" here, but I think the ability to instantly plop down a rectangle in the middle of the screen and then drag center/vertices around is pretty nice.  Trace/Draw type function is also nice, but it is a little more complicated with what mode are you in (tracing or not tracing).

>     * It your proposition can the 'File' button from the toolbar be simplified as the "import flight plan" is on the main popup ?

Maybe, although some people might want to "open/import" a saved mission right from the File menu instead of having to press "New" first.

>     * Can't we simplify "Create Plan" to 1 level by adding "Polygon Survey", "Circle Survey"  as it occupies all the screen and appears to be half empty ?

Maybe, but what about Corridor Scan and Structure Scan and other future mission types that support KML/SHP inputs?  Things could get messy.  Also, your screen size really affects how many plan types are shown.  Some visual clean up is needed either way.

> Can't "Open Saved Mission" and Import from KML be merged to an "Import" button ?

Not really.  KML/SHP can be used for surveys, corridor scans, and structure scans.  How would you know which you are creating from the KML/SHP?

> From the other image image, what do you mean by "Use the insert function on the right hand list ?
> You mean this submenu ?
> [image]
>
> Because this is not practical at all as it adds a MissionItem [...]

huh... is this a bug?

tilaktilak, 2019-10-28:
> huh... is this a bug?
Not really as it's an "insert" button, it does not append the list but insert the WP between 2 items. It's not intended to be used to create your mission, but more likely to add custom mission item (like a camera type, ...).

Antiheavy, 2019-10-28:
> "Import" appears on the popup wizard window

> Couldn't it be done by the Import window ?

I don't see the word "Import" anywhere in the UI on recent QGC daily builds.  Am I missing something?  Or are you proposing a new button option?

> I really think that the "basic" method has became redundant/obsolete since we are able to trace polygons. On a touchscreen it's easier to trace a polygon than moving points, I would rather trace a polygon and minimize the "adjusting" part

I've used a lot of different ground station applications and some use the trace/draw style of area creation and some use the auto-create rectangle and drag corners style.   It comes down to personal preference and use cases and from my experience is a 50/50 split which style users like better.  If the "Polygon Tools" menu stays, I think people would like the ability to choose which option of polygon creation they like.

tilaktilak, 2019-10-28:
> I don't see the word "Import" anywhere in the UI on recent QGC daily builds. Am I missing something? Or are you proposing a new button option?

I'm referring to your screenshot, it's "Open Saved Mission" or "Create from KML/SHP".

DonLakeFlyer, 2019-11-11:
> * Can't "Open Saved Mission" and Import from KML be merged to an "Import" button ?

File/Open already works like that. You can open .plan, .kml or .shp file from there. If you open a .kml or .shp and the shape is a polygon and you are flying multi-rotor it prompts you as to whether you want create a Survey or a Structure Scan.

DonLakeFlyer, 2019-11-11:
FYI: Working my way towards a much more contextual left hand set of tools like @Antiheavy has been talking about. As well as a new Takeoff pattern for fixed wing.

DonLakeFlyer, 2019-12-11:
This is mostly done with all the new stuff. Any new problems can't be dealt with in new issues

closed 2019-12-11 by DonLakeFlyer