mavlink/qgroundcontrol #7909: Pixhawk cant connect to any computer

themystick, 2019-10-11:
Hi. I'm using Pixhawk 2.1 Cube. I have a problem with my device and I've tried to solve this problem many times, I've tried many things but still not working. So I need your help. The problem is my device is not connecting to any computer. When I tried to connect via micro usb port, the led (yellow and slowly blinked amber) is on but I can't see the driver in the device manager, I mean there is no difference in the device manager. So I've tried to connect via RFD and I can't connect again. (I've tried this with other computers). 

So interestingly, I've tried to connect with "debug" port and I've seen that the COM3 port is now available in the  device manager but the leds are off. Just connecting but It seems there is no power in the pixhawk. So I've tried to connect MP again and it's not working. 

So I've tried everything but It's not working. Please help me about this issue. Btw I've upload the video about the problem If you want . Thanks. Regards

DAMO238, 2019-10-11:
What OS are you running QGC from? If Windows, make sure you have the UAV drivers installed, or Windows will not recognise it but will still supply power.

themystick, 2019-10-12:
I've tried with Windows and Ubuntu many times. Still nothing work.

DonLakeFlyer, 2019-10-15:
You should ask for help in the CubePilot.org forums. This sounds like more of a hardware problem than a ground station issue.

closed 2019-10-15 by DonLakeFlyer