mavlink/qgroundcontrol #7917: RSSI indicator is showing random value

zwn, 2019-10-15:
I have pixhawk 4 running px4 1.9.2 and qgroundcontrol 3.5.4 connected over usb. Pixhawk is connected to FrSky RX6R receiver using SBUS. I am using X-Lite S transmitter. 

AFAIK there is no datasource for the RSSI value but the indicator in the top toolbar is showing RSSI value 39% - mostly one bar but sometimes two bars filled - when value changes to 40%. That is with the transmitter turned off. When I turn it on the value does not change. 

I tried to set RC_RSSI_PWM_CHAN to point to a channel where I have published rssi which results in some change of the indicator. However I am not able to related the shown values with what my transmitter is reporting, not even basic up & down movement (72dB -> 93%, 108dB -> 91%). But qgroundcontrol is able to detect when I turn on and off the transmitter (reporting manual control lost or regained).

There is no documentation about how to send rssi value to the channel (mixer setup) to be properly understood by px4 and/or qgroundcontrol. I was not able to find any tool inside qgroundcontrol that would show me the raw numbers that correspond to the white circles in Channel Monitor view so I am not able to provide more info. I have also not found any documentation describing what the indicator is supposed to be indicating.

I am on ubuntu if that means anything.

zwn, 2019-10-16:
Using https://oscarliang.com/rssi-ppm-channel-taranis/ I have been able to set up RSSI to be output as one of the channels. In addition to that I had to go to Outputs setup in OpenTx and invert the channel. When RC_RSSI_PWM_CHAN is set the the corresponding channel, the RSSI indicator seems to be somewhat following - goes up in % when the signal improves.

Even thought I got it working, there are some issues:
* the indicator is showing random value when RC_RSSI_PWM_CHAN is undefined
* I couldn't find anything in documentation that would tell me what the RSSI indicator is supposed to be showing (manual radio rssi, telemetry link rssi, where it gets the info from, is it displayed raw with "%" appended or is there any transformation applied etc.)

DonLakeFlyer, 2019-10-16:
@dogmaphobic ?

dogmaphobic, 2019-12-26:
It seems any RSSI other than the one from a SiK radio is not handled at all. The code checks to see if the message is from a SiK radio. if so, it does some conversion here:

https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L1748-L1760

For anything else, the value is stored as is. 

The [message definition](https://mavlink.io/en/messages/common.html#RADIO_STATUS) now says that the RSSI value is "device dependent" and ranges from 0 to 254. A value of 255 indicates the lack of RSSI. 

If I change the code to conform to the new message definition, how do we handle the SiK radio? Even then, what do I make out of "device dependent"? Just normalize 0-254 to 0-100?

DonLakeFlyer, 2019-12-27:
Might take on this is that given the current state of RADIO_STATUS you can't do anything other show a raw value for RSSI. You can' translate it into some pretty visual and/or to a percentage.

@hamishwillee I'd suggest a discussion for the mavlink committee is to deprecate RADIO_STATUS since it's useless and make a brand new message which is spec'ed and implemented in a single way from the start.

hamishwillee, 2020-01-06:
@dogmaphobic We, including you, discussed rssi extensively in https://github.com/mavlink/mavlink/issues/1027
I begged for comments on the wording! 

My original understanding was that this would be a linear scale (%) with boundaries from 0-254 (rather than 0-100 to increase resolution). At some point Edwin stated that this should be device independent and burden of scaling should go onto QGC, - but we didn't resolve finding out what that scaling should be.

Right now I don't know what I was thinking - obviously it makes more sense for the device to provide either an absolute value or a percentage. Forcing every other system to know what it is talking to is nuts.

One option is to change this again to what most people probably expected "Receive signal strength indicator. Values: [0-254] (linearly increasing), 255: invalid/unknown." I think this is what ArduPilot does.

Otherwise, please state what you want exactly and I can work with the mavlink team to facilitate.

hamishwillee, 2020-01-09:
@DonLakeFlyer Had a chat in the dev meeting regarding RSSI. The summary is that to be usefully presented by GCS either the information has to include the rssi units/scale or it has to be in a consistent unit/scale. Alternatively the GCS might not even care about the scale and just want a level indicator "good/bad" or "1-5". 

Probably best way to progress this is for you guys to propose how you want it to be in a PR and we then discuss. Unless there is a real stakeholder driving and implementing things this is all very abstract.

dogmaphobic, 2020-01-09:
> Alternatively the GCS might not even care about the scale and just want a level indicator "good/bad" or "1-5".

The problem is how to I convert arbitrary ranges into 1-5? The SiK radio sends some proprietary data that must be translated into dBm (the only thing currently supported). The specs used to call for 0-100 and having 255 as a flag for "none". Then we got this new thing about having 0-254 while 255 flags it as "none". In my opinion, I would force everything to be 0-100 as it involves no transformation within the GS. The 0-254 might give you a bit more granularity but I think it's meaningless in this context, only making something simple into something more complicated than it needs to be.

closed 2026-10-04 by github-actions[bot]