mavlink/qgroundcontrol #7920: Map provider broken daily?

MaEtUgR, 2019-10-16:
## Expected Behavior
QGC shows the map as usual, either there's a default map that has world wide coverage or there's a prompt to choose one to begin with.

## Current Behavior
I first notices the whole map view just being white 2 days ago on daily. Now I realized with that older version or todays daily it's basically random what map I get on any boot. And the map provider is always unselected in the settings:

## Log Files and Screenshots
[image]
[image]
[image]
[image]
[image]

FYI @MatejFranceskin @DanielePettenuzzo

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10 and Manjaro 18 (up to date)
- QGC Version: 
[image]
- Flight Controller: None/PX4 master SITL

DonLakeFlyer, 2019-10-16:
@tilaktilak Is working on it

closed 2019-10-16 by DonLakeFlyer

MaEtUgR, 2019-10-17:
Thanks, it works again :+1: