mavlink/qgroundcontrol #7922: MacOS: Master crash on start up

FlavioTonelli, 2019-10-16:
I'm compiling and running the current master of QGC on macOS Catalina and Qt 5.11.0
When I start the application I always experience a crash.
Stable Commit 9b46aacac0bc2c13d4c39ffddb96ed80548328ad is working well.

```
process:               QGroundControl [38474]

Path:                  /Users/USER/*/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               0.0.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [29652]
Responsible:           Qt Creator [29652]
User ID:               501

Date/Time:             2019-10-16 19:53:19.385 +0200
OS Version:            Mac OS X 10.15 (19A602)
Report Version:        12
Anonymous UUID:        83C7260C-991C-AB6E-744E-3E086DFA0D45

Sleep/Wake UUID:       15EF4D5E-D2C3-40D8-8F4C-48F8CBD0FBA0

Time Awake Since Boot: 15000 seconds
Time Since Wake:       7800 seconds

System Integrity Protection: enabled

Crashed Thread:        14  QSGRenderThread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [38474]

Application Specific Information:
-[NSOpenGLContext setView:] must be called from the main thread.

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6c9e68f6 __psynch_cvwait + 10
1   libsystem_pthread.dylib       	0x00007fff6caa7082 _pthread_cond_wait + 701
2   org.qt-project.QtCore         	0x000000011050cc7b QWaitConditionPrivate::wait(unsigned long) + 75
3   org.qt-project.QtCore         	0x000000011050cc12 QWaitCondition::wait(QMutex*, unsigned long) + 162
4   org.qt-project.QtQuick        	0x000000011288cc99 QSGThreadedRenderLoop::polishAndSync(QSGThreadedRenderLoop::Window*, bool) + 1481
5   org.qt-project.QtQuick        	0x000000011288c4ad QSGThreadedRenderLoop::handleExposure(QQuickWindow*) + 1117
6   org.qt-project.QtQuick        	0x000000011288bfe9 QSGThreadedRenderLoop::exposureChanged(QQuickWindow*) + 281
7   org.qt-project.QtGui          	0x0000000112f54cf1 QWindow::event(QEvent*) + 817
8   org.qt-project.QtQuick        	0x00000001128e4fd7 QQuickWindow::event(QEvent*) + 263
9   org.qt-project.QtCore         	0x00000001106bc3d8 QCoreApplicationPrivate::notify_helper(QObject*, QEvent*) + 88
10  org.qt-project.QtCore         	0x00000001106bc0ed QCoreApplication::notify(QObject*, QEvent*) + 45
11  org.qt-project.QtCore         	0x00000001106bc03f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
12  org.qt-project.QtGui          	0x0000000112f48f34 QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) + 308
13  org.qt-project.QtGui          	0x0000000112f26c03 bool QWindowSystemInterfacePrivate::handleWindowSystemEvent<QWindowSystemInterface::SynchronousDelivery>(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 99
14  org.qt-project.QtGui          	0x0000000112f2cea0 void QWindowSystemInterface::handleExposeEvent<QWindowSystemInterface::SynchronousDelivery>(QWindow*, QRegion const&) + 192
15  libqcocoa.dylib               	0x0000000116cbd41f QCocoaWindow::handleExposeEvent(QRegion const&) + 463
16  libqcocoa.dylib               	0x0000000116cc3880 -[QNSView updateRegion:] + 320
17  libqcocoa.dylib               	0x0000000116cc3ced -[QNSView updateLayer] + 685
18  com.apple.AppKit              	0x00007fff328b1edf _NSViewUpdateLayer + 98
19  com.apple.AppKit              	0x00007fff328b1855 -[_NSViewBackingLayer display] + 501
20  com.apple.QuartzCore          	0x00007fff40dae0c1 CA::Layer::display_if_needed(CA::Transaction*) + 739
21  com.apple.QuartzCore          	0x00007fff40ed2821 CA::Context::commit_transaction(CA::Transaction*, double) + 341
22  com.apple.QuartzCore          	0x00007fff40d9cec6 CA::Transaction::commit() + 638
23  com.apple.AppKit              	0x00007fff3307ce2a __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 266
24  com.apple.AppKit              	0x00007fff3307d252 ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
25  com.apple.CoreFoundation      	0x00007fff355f84be __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
26  com.apple.CoreFoundation      	0x00007fff355f83e4 __CFRunLoopDoObservers + 457
27  com.apple.CoreFoundation      	0x00007fff3559b55b __CFRunLoopRun + 1219
28  com.apple.CoreFoundation      	0x00007fff3559ae13 CFRunLoopRunSpecific + 499
29  org.libsdl.SDL2               	0x0000000110207dd2 0x110192000 + 482770
30  org.libsdl.SDL2               	0x000000011022328a 0x110192000 + 594570
31  org.libsdl.SDL2               	0x00000001102239c8 0x110192000 + 596424
32  org.libsdl.SDL2               	0x00000001102238ba 0x110192000 + 596154
33  org.libsdl.SDL2               	0x00000001101ab02f 0x110192000 + 102447
34  org.libsdl.SDL2               	0x00000001101aee94 0x110192000 + 118420
35  org.qgroundcontrol.qgroundcontrol	0x000000010e383902 JoystickSDL::init() + 18 (JoystickSDL.cc:17)
36  org.qgroundcontrol.qgroundcontrol	0x000000010e1382d9 JoystickManager::init() + 25 (JoystickManager.cc:57)
37  org.qgroundcontrol.qgroundcontrol	0x000000010e22f95f QGCApplication::_initForNormalAppBoot() + 1023 (QGCApplication.cc:571)
38  org.qgroundcontrol.qgroundcontrol	0x000000010e32ab2b main + 2187 (main.cc:390)
39  org.qgroundcontrol.qgroundcontrol	0x000000010e01e304 start + 52
```
[....]

DonLakeFlyer, 2019-10-16:
First thing to try is to use 5.11.3 which is the official version. Could just be some Catalina specific problem though with 5.11. If 5.11.3 doesn't work. Could you try with the latest 5.12? We are moving to that.

FlavioTonelli, 2019-10-16:
Tested both 5.11.3 and 5.12 as suggested.
QGC is correctly running only on 5.12

DonLakeFlyer, 2019-10-16:
Ok, then it's a problem that 5.11 doesn't support Catalina.

@stefandunca This is going to need to verified as fixed once the move to 5.12 is done.

DonLakeFlyer, 2019-10-16:
@FlavioTonelli You should be ok building with 5.12. Both Gus and I have been using 5.12 for a while now.

FlavioTonelli, 2019-10-16:
@DonLakeFlyer thank you for your feedback.
I'm going to test the macOS 5.12 version during the next days :) I will report back in case of issues.

DonLakeFlyer, 2019-12-11:
Move to 5.12 is specified in another issue for Stable

closed 2019-12-11 by DonLakeFlyer