mavlink/qgroundcontrol #7932: UAV Drivers Package and Old Drivers being installed with current daily

mwiatt, 2019-10-18:
From https://github.com/mavlink/qgroundcontrol/issues/7523#issuecomment-543862233

From @mwiatt
> I uninstalled QGC, UAV Drivers, and 21x Windows Driver Package XXX. Then installed daily. I ended up with UAV Drivers + all 21x Windows Driver Package XXX.

From @DonLakeFlyer 
> Hmm. That doesn't seem correct. Can you create a new issue with details and we can work from there.

## Steps to Reproduce:
1. DL new daily from https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl-installer.exe
2. Install

## System Information
- Operating System: Windows 7 and 10
- QGC build: daily

## Log Files and Screenshots
Confirmed in Win 7 and 10. Below series of screenshots is from Win7:  
[image]

DonLakeFlyer, 2019-10-18:
The setup screenshots are correct behavior. Except for the end result in control panel.

mwiatt, 2019-10-18:
I installed with Win10 with the same behavior. These systems have had an old stable version installed. Perhaps the driver installer is an old cached version?

DonLakeFlyer, 2019-10-18:
> These systems have had an old stable version installed.

Old stable may have looked like that I think. The driver install is a bit messy. Only cleaned up in a way that is manageable with the latest set. By that I mean playing nice with MP as well as uninstall/update. So if Win 10 had old build on it those drivers could have been there before. I started from uninstalling all drivers and then reinstalled daily.

mwiatt, 2019-10-18:
I should have been more clear...

Win10 system had old stable version 3.5.4. I uninstalled QGC, UAV Drivers, and the 21 Windows Driver Packages. Then installed daily and ended up with all 21 Windows Driver Packages. Was wondering if possibly the driver installer from 3.5.4 was cached somewhere and not being overwritten when the new daily installer was unpacking.

DonLakeFlyer, 2019-10-18:
> Was wondering if possibly the driver installer from 3.5.4 was cached somewhere and not being overwritten when the new daily installer was unpacking.

Maybe. Let me try that sequence.

DonLakeFlyer, 2020-02-08:
I futzed around with this and it seems to work correctly. I can believe that somewhere along the way it was broken. And if you update from that specific build it would be wrong. But there isn't any way to go back in time and fix that.

closed 2020-02-08 by DonLakeFlyer