mavlink/qgroundcontrol #7935: Firmware Upgrade one MacOS 10.15 Catalina not working

themcfadden, 2019-10-18:
## Current Behavior
Using QGC v3.5.5 and v3.5.4, macOS Catalina (10.15), Firmware upgrade does not work. After unplugging Pixhawk from USB, firmware upgrade does not start.

## Steps to Reproduce:
1. In QGC v3.5.5 or v3.5.4, plug your PixHawk into a USB port and establish a connection.
2. Navigate to the Firmware Setup page and follow instructions and unplug the PixHawk to start the firmware upgrade process.
3. Follow instructions and plug the PixHawk back into the USB port.
4. Nothing happens as it is supposed to start the firmware upgrade process.

## System Information
- Operating System: macOS Catalina v10.15
- QGC Version: 3.5.4 or 3.5.5
- QGC build: Official release downloaded from links in docs.
- Flight Controller: Pixhawk 4
- Autopilot (with version): PX4 1.9.2

## Detailed Description
It has crashed on this screen a few times on me too, but I think it is known that the MacOS version is not very stable.

DonLakeFlyer, 2019-10-19:
Is this a problem only with Stable. Does a daily build work?

> It has crashed on this screen a few times on me too, but I think it is known that the MacOS version is not very stable.

QGC should never crash. News to me that the Mac version is unstable.

themcfadden, 2019-10-21:
Thank you for the quick response.

I got the daily build to work - it has more information on the Firmware setup page. It shows the detected devices, which is very helpful. It looks like the FTDI cable I'm using to monitor the PX4 console looks like a Sik Telemetry radio and is confusing the system. Unplugging it made the firmware upgrade work as expected.

I'll submit an issue the next time MacOS QGC crashes on me. There are a couple of us here that have experienced this sort of thing.

DonLakeFlyer, 2019-10-21:
> I'll submit an issue the next time MacOS QGC crashes on me.

I vaguely remember daily build crash fixes for firmware upgrade. Daily build firmware upgrade for ArduPilot is significantly different than Stable with respect to board detection, ArduPilot bootloader support and picking the right firmware based on board type.

closed 2019-10-21 by DonLakeFlyer