mavlink/qgroundcontrol #7942: MacOS Crash on Firmware Upgrade

themcfadden, 2019-10-21:
## Current Behavior
QGC crashed while attempting a firmware upgrade. When I unplugged the USB as instructed, QGC crashed.

I'll switch to the daily build and see if it still happens.

## Crash Log

Process:               QGroundControl [92720]
Path:                  /Applications/QGroundControl 3.5.5.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               3.5.5 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QGroundControl [92720]
User ID:               501

Date/Time:             2019-10-21 15:38:47.103 -0600
OS Version:            Mac OS X 10.15 (19A602)
Report Version:        12
Bridge OS Version:     3.0 (14Y906)
Anonymous UUID:        E88215AF-63C2-BD0C-1808-07BE782B42A5

Time Awake Since Boot: 24000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [92720]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qgroundcontrol.qgroundcontrol	0x000000010cc74681 0x10cb91000 + 931457
1   org.qgroundcontrol.qgroundcontrol	0x000000010cc73168 0x10cb91000 + 926056
2   org.qt-project.QtCore         	0x000000010e0bb02b QMetaObject::activate(QObject*, int, int, void**) + 2347
3   org.qt-project.QtCore         	0x000000010e0c2c22 QTimer::timerEvent(QTimerEvent*) + 98
4   org.qt-project.QtCore         	0x000000010e0b31a6 QObject::event(QEvent*) + 102
5   org.qt-project.QtWidgets      	0x000000010f221ee2 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 306
6   org.qt-project.QtWidgets      	0x000000010f2231ff QApplication::notify(QObject*, QEvent*) + 383
7   org.qt-project.QtCore         	0x000000010e08a03f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
8   org.qt-project.QtCore         	0x000000010e0e18da QTimerInfoList::activateTimers() + 890
9   libqcocoa.dylib               	0x00000001123743e2 0x11234b000 + 168930
10  com.apple.CoreFoundation      	0x00007fff3157da21 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
11  com.apple.CoreFoundation      	0x00007fff3157d9c0 __CFRunLoopDoSource0 + 103
12  com.apple.CoreFoundation      	0x00007fff31560f84 __CFRunLoopDoSources0 + 209
13  com.apple.CoreFoundation      	0x00007fff31560590 __CFRunLoopRun + 1272
14  com.apple.CoreFoundation      	0x00007fff3155fe13 CFRunLoopRunSpecific + 499
15  com.apple.HIToolbox           	0x00007fff300ecb2d RunCurrentEventLoopInMode + 292
16  com.apple.HIToolbox           	0x00007fff300ec86d ReceiveNextEventCommon + 600
17  com.apple.HIToolbox           	0x00007fff300ec5f7 _BlockUntilNextEventMatchingListInModeWithFilter + 64
18  com.apple.AppKit              	0x00007fff2e782ac4 _DPSNextEvent + 990
19  com.apple.AppKit              	0x00007fff2e781834 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
20  com.apple.AppKit              	0x00007fff2e77bfd4 -[NSApplication run] + 658
21  libqcocoa.dylib               	0x00000001123750bd 0x11234b000 + 172221
22  org.qt-project.QtCore         	0x000000010e085b92 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 418
23  org.qt-project.QtCore         	0x000000010e08a752 QCoreApplication::exec() + 402
24  org.qgroundcontrol.qgroundcontrol	0x000000010cd301ab 0x10cb91000 + 1700267
25  org.qgroundcontrol.qgroundcontrol	0x000000010cb9a174 start + 52

Thread 1:: AMCP Logging Spool
0   libsystem_kernel.dylib        	0x00007fff6880f182 semaphore_wait_trap + 10
1   com.apple.audio.caulk         	0x00007fff648e9ec6 caulk::mach::semaphore::wait() + 16
2   com.apple.audio.caulk         	0x00007fff648e5442 caulk::semaphore::timed_wait(double) + 106
3   com.apple.audio.caulk         	0x00007fff648e8a14 caulk::concurrent::details::worker_thread::run() + 30
4   com.apple.audio.caulk         	0x00007fff648e8b64 void* caulk::thread_proxy<std::__1::tuple<caulk::thread::attributes, void (caulk::concurrent::details::worker_thread::*)(), std::__1::tuple<caulk::concurrent::details::worker_thread*> > >(void*) + 45
[... 564 lines cut ...]
TOTAL, minus reserved VM space     1.3G     1845 

Model: MacBookPro13,3, BootROM 262.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2.6 GHz, 16 GB, SMC 2.38f8
Graphics: kHW_IntelHDGraphics530Item, Intel HD Graphics 530, spdisplays_builtin
Graphics: kHW_AMDRadeonPro450Item, AMD Radeon Pro 450, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.77.105.1 AirPortDriverBrcmNIC-1429)
Bluetooth: Version 7.0.0f8, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet Slot  1, Ethernet, en7
PCI Card: pci1b73,1100, sppci_usbxhci, Thunderbolt@196,0,0
PCI Card: pci1b73,1100, sppci_usbxhci, Thunderbolt@197,0,0
PCI Card: ethernet, sppci_ethernet, Thunderbolt@195,0,0
USB Device: USB 3.0 Bus
USB Device: Apple T1 Controller
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB2.0 Hub
USB Device: FT232R USB UART
USB Device: USB Advanced Audio Device
USB Device: USB 3.0 Bus
USB Device: USB Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Device: TB3DOCK2DPPD, StarTech.com, 3, 23.1

themcfadden, 2019-10-21:
Just had another crash after completing calibration. QGC 3.5.5 still. I saved the crash log if you are interested.

DonLakeFlyer, 2019-10-22:
Yeah can you put that one in here as well. Then I can see if they are already taken care of in daily.

themcfadden, 2019-10-22:
I believe this one occurred after completing a calibration.

Process:               QGroundControl [92847]
Path:                  /Applications/QGroundControl 3.5.5.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               3.5.5 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QGroundControl [92847]
User ID:               501

Date/Time:             2019-10-21 15:55:11.110 -0600
OS Version:            Mac OS X 10.15 (19A602)
Report Version:        12
Bridge OS Version:     3.0 (14Y906)
Anonymous UUID:        E88215AF-63C2-BD0C-1808-07BE782B42A5

Time Awake Since Boot: 25000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000078
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [92847]

VM Regions Near 0x78:
--> 
    __TEXT                 00000001006af000-0000000101519000 [ 14.4M] r-x/rwx SM=COW  /Applications/QGroundControl 3.5.5.app/Contents/MacOS/QGroundControl

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qgroundcontrol.qgroundcontrol	0x0000000100792681 0x1006af000 + 931457
1   org.qgroundcontrol.qgroundcontrol	0x0000000100791168 0x1006af000 + 926056
2   org.qt-project.QtCore         	0x0000000101bc502b QMetaObject::activate(QObject*, int, int, void**) + 2347
3   org.qt-project.QtCore         	0x0000000101bccc22 QTimer::timerEvent(QTimerEvent*) + 98
4   org.qt-project.QtCore         	0x0000000101bbd1a6 QObject::event(QEvent*) + 102
5   org.qt-project.QtWidgets      	0x0000000102d20ee2 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 306
6   org.qt-project.QtWidgets      	0x0000000102d221ff QApplication::notify(QObject*, QEvent*) + 383
7   org.qt-project.QtCore         	0x0000000101b9403f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
8   org.qt-project.QtCore         	0x0000000101beb8da QTimerInfoList::activateTimers() + 890
9   libqcocoa.dylib               	0x0000000105e113e2 0x105de8000 + 168930
10  com.apple.CoreFoundation      	0x00007fff3157da21 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
11  com.apple.CoreFoundation      	0x00007fff3157d9c0 __CFRunLoopDoSource0 + 103
12  com.apple.CoreFoundation      	0x00007fff31560f84 __CFRunLoopDoSources0 + 209
13  com.apple.CoreFoundation      	0x00007fff31560590 __CFRunLoopRun + 1272
14  com.apple.CoreFoundation      	0x00007fff3155fe13 CFRunLoopRunSpecific + 499
15  com.apple.HIToolbox           	0x00007fff300ecb2d RunCurrentEventLoopInMode + 292
16  com.apple.HIToolbox           	0x00007fff300ec86d ReceiveNextEventCommon + 600
17  com.apple.HIToolbox           	0x00007fff300ec5f7 _BlockUntilNextEventMatchingListInModeWithFilter + 64
18  com.apple.AppKit              	0x00007fff2e782ac4 _DPSNextEvent + 990
19  com.apple.AppKit              	0x00007fff2e781834 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
20  com.apple.AppKit              	0x00007fff2e77bfd4 -[NSApplication run] + 658
21  libqcocoa.dylib               	0x0000000105e120bd 0x105de8000 + 172221
22  org.qt-project.QtCore         	0x0000000101b8fb92 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 418
23  org.qt-project.QtCore         	0x0000000101b94752 QCoreApplication::exec() + 402
24  org.qgroundcontrol.qgroundcontrol	0x000000010084e1ab 0x1006af000 + 1700267
25  org.qgroundcontrol.qgroundcontrol	0x00000001006b8174 start + 52

Thread 1:: AMCP Logging Spool
0   libsystem_kernel.dylib        	0x00007fff6880f182 semaphore_wait_trap + 10
1   com.apple.audio.caulk         	0x00007fff648e9ec6 caulk::mach::semaphore::wait() + 16
2   com.apple.audio.caulk         	0x00007fff648e5442 caulk::semaphore::timed_wait(double) + 106
3   com.apple.audio.caulk         	0x00007fff648e8a14 caulk::concurrent::details::worker_thread::run() + 30
4   com.apple.audio.caulk         	0x00007fff648e8b64 void* caulk::thread_proxy<std::__1::tuple<caulk::thread::attributes, void (caulk::concurrent::details::worker_thread::*)(), std::__1::tuple<caulk::concurrent::details::worker_thread*> > >(void*) + 45
5   libsystem_pthread.dylib       	0x00007fff688d2d76 _pthread_start + 125
[... 549 lines cut ...]
TOTAL                              1.6G     1883 
TOTAL, minus reserved VM space     1.3G     1883 

Model: MacBookPro13,3, BootROM 262.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2.6 GHz, 16 GB, SMC 2.38f8
Graphics: kHW_IntelHDGraphics530Item, Intel HD Graphics 530, spdisplays_builtin
Graphics: kHW_AMDRadeonPro450Item, AMD Radeon Pro 450, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.77.105.1 AirPortDriverBrcmNIC-1429)
Bluetooth: Version 7.0.0f8, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet Slot  1, Ethernet, en7
PCI Card: pci1b73,1100, sppci_usbxhci, Thunderbolt@196,0,0
PCI Card: pci1b73,1100, sppci_usbxhci, Thunderbolt@197,0,0
PCI Card: ethernet, sppci_ethernet, Thunderbolt@195,0,0
USB Device: USB 3.0 Bus
USB Device: Apple T1 Controller
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB2.0 Hub
USB Device: USB Advanced Audio Device
USB Device: USB 3.0 Bus
USB Device: USB Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Device: TB3DOCK2DPPD, StarTech.com, 3, 23.1

DonLakeFlyer, 2019-10-22:
Thanks. The crash looks to be the same in both cases. @dogmaphobic Do you remember anything about a Timer referencing a null object in Stable that was fixed. Doesn't ring a bell.

DonLakeFlyer, 2019-10-22:
Looking more might not be a pull pointer.

DonLakeFlyer, 2019-10-22:
> I believe this one occurred after completing a calibration.

Did you tell QGC to reboot the vehicle when that happened?

DonLakeFlyer, 2019-10-22:
Unfortunately the stack trace is pretty generic with respect to not being helpful to track down the actual problem. Going to need some more concrete steps to reproduce. I tried the firmware upgrade unplug thing myself and of course it all works fine.

themcfadden, 2019-10-22:
> > I believe this one occurred after completing a calibration.
> 
> Did you tell QGC to reboot the vehicle when that happened?

Probably not.

themcfadden, 2019-10-22:
> Unfortunately the stack trace is pretty generic with respect to not being helpful to track down the actual problem. Going to need some more concrete steps to reproduce. I tried the firmware upgrade unplug thing myself and of course it all works fine.

I'll spend some time on it, but I think that will be tough. My impression is that it has a good bit of randomness to it.

I've noticed crashes during firmware update, and doing lots of connect/disconnect (due to resetting drone). It has also died a couple times while flying mission - again I think associated with losing/regaining connection.

DonLakeFlyer, 2024-10-13:
Closing assuming this is all different now. Lot's of work went into LinkManager which fixed a bunch of crashes. And the current daily build is having of work put into it to fix up remaining threading problems. So hopefully this is gone now.

closed 2024-10-13 by DonLakeFlyer