mavlink/qgroundcontrol #7950: Incorrect msg that "You are using a version prior to 3.3" with ArduPilot

rmackay9, 2019-10-26:
I've loaded Copter's latest beta version (4.0.0-rc1) to my multicopter but when I connect with QGC it incorrectly displays a warning that I'm using a version prior to 3.3.  Maybe it's confused by our recent jump from 3.6.x to 4.0.x?

[image]

DonLakeFlyer, 2019-11-09:
Fixed in daily builds. Stable is also fixed but a new point release hasn't been pushed out yet.

closed 2019-11-09 by DonLakeFlyer

rmackay9, 2019-11-11:
Great, thanks for this!

Cattlebaron, 2020-05-02:
I have upgraded to the latest version I think it maybe version 4.0.5 based on something that flashed across the screen but who knows. Subsequently I flashed the latest release of the firmware onto my Quadcopter and followed the setup routine. This was problem free. When I came to use my iPad to control the quad I got the dreaded message "QGroundControl fully supports Version 3.3 and above. You are using a version prior to that.” 
The question is the quad useable with my TX and forget about the iPad and 'missions' or do I pack the quad away until a new version of GQC is available for my iPad?

DonLakeFlyer, 2020-05-03:
The iOS build hasn't been updated yet with latest Stable so you will still see this bug. It's fine to ignore it.