mavlink/qgroundcontrol #7954: Can't compile QGC, fatal error LNK1104: cannot open file 'z.lib'

olliw42, 2019-10-28:
I would very much appreciate help with this issue which started to occur some while ago (few months). I can't distinguish if it is just me or if it indeed points to some general issue, but I somewhat suspect the latter and hence am posting here. 

The issue: Since a while I can't build QGroundControl anymore, building with QtCreator on Windows fails with: 

> \Users\xxxxxx\AppData\Local\Temp\QGroundControl.exe.1784.176952.jom
> LINK : fatal error LNK1104: cannot open file 'z.lib'
> jom: D:\private\Git\build-qgroundcontrol-Desktop_Qt_5_11_3_MSVC2015_32bit-Debug\Makefile.Debug [debug\QGroundControl.exe] Error 1104
> jom: D:\private\Git\build-qgroundcontrol-Desktop_Qt_5_11_3_MSVC2015_32bit-Debug\Makefile [debug] Error 2
> 10:01:20: The process "C:\Qt\Tools\QtCreator\bin\jom.exe" exited with code 2.
> Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.11.3 MSVC2015 32bit)
> When executing step "Make"
> 10:01:20: Elapsed time: 03:05.

I did the set up as described in the manual: https://dev.qgroundcontrol.com/en/getting_started/ for Windows on two computers running Win7.

Notes:
* I observe the very same behavior on two different computers. Especially the result on the 2nd computer is noteworthy, since here I didn't had QGC installed before, i.e., it should be a "fresh" and "clean" install.
* De-install, re-install, deleting build folders, etc, I all tried, and nothing I tried helped.
* I didn't had this issue before. That is, I had a working install on the 1st computer which allowed me to build and run QGC. The only thing I'm aware of which I did was to upgrade the git repository and from a certain point on the issue occurred (that is, I did upgrade the git repo several times also before but this always worked) . I can't remember the exact date anymore when this issue started to occur, but it was around August. In Juli I did this PR, #7606, and at this time I for sure had a working setup and could build. (it thus should not be issues like spaces in path and so on!)

I did a web research, but couldn't find anything which would have allowed me to identify or resolve the issue (which doesn't mean much, though, only that I did an effort). 

It seems I'm not the only one: https://discuss.px4.io/t/when-iam-trying-to-compile-qgroundcontrol-pro-in-qtcreator-in-windows/13502. Also google gives some gitter links. These links don't seem to provide a solution either.

Since this seemed to happen around the time where you switched to QtQuick and other such "deep" changes, I somehow suspect that the issue might be due to a change in the build system which hasn't yet found it's way into the documentation. But that's mere speculation.

Any help or pointers would be much appreciated.

Thx, Olli

tilaktilak, 2019-10-28:
This error means that the linker cannot "open" zlib, not sure if it's because it's not found.   
I'm not working on Windows, but by having a look at [QGCExternalLibs.pri](https://github.com/mavlink/qgroundcontrol/blob/31d8c52ec777c7ae84e3cf9e9c3d01731f742ca3/QGCExternalLibs.pri#L116) I can tell that the lib is supposed to be in the subfolder libs/zlib/Windows/

So you can check if `-spec win32msvc` is part of your qmake command options and verify if INCLUDE_PATH and LIBS both contains qgroundcontrol/libs/zlib/Windows/include and libs.

I hope it will help.

olliw42, 2019-10-29:
thx for the response

I do have files zlib.h and zlib.lib in my folders `.../Documents/GitHub/qgroundcontrol/libs/zlib/windows/include` and `.../Documents/GitHub/qgroundcontrol/libs/zlib/windows/lib`, respectively. I do not seem to have a file `z.lib`, at least searching my C drive couldn't find it.

I can find a MakeFile file and a cmake folder. I also can find a file `.qmake.stash` in the `build-qgroundcontrol-Desktop_Qt_5_11_0_MSVC2015_32bit-Debug` folder, but nothing else with qmake in its name. In that folder there is also a `Makefile.Debug` file. 

I wouldn't know how to check for the option you mentioned.

As regards `INCLUDE_PATH `and `LIBS`. From the above mentioned text files I find that `Makefile.Debug` has a `INCPATH `entry, which among others has entries `-IC:\Users\Olli\Documents\GitHub\qgroundcontrol\libs\zlib\Windows\include` and `-I\include -IC:\Qt\5.11.0\msvc2015\mkspecs\win32-msvcm`, and that it has a `LIBS `entry which has an entry `/LIBPATH:C:\Users\Olli\Documents\GitHub\qgroundcontrol\libs\zlib\Windows\libs`

I maybe should add: I'm notoriously a nut when it comes to build systems (I kind of expect them to work after install and never spend time on them). Sorry for that.

I would not be surprised if at the end it would be just a simple thing like some files or folders renamed or relocated.

olliw42, 2019-10-29:
I did a quick diff between master 7.Aug.2019 and master 25.Oct.2019 for the file  QGCExternalLibs.pri, and it seems that the INCLUDE and LIB paths have indeed been changed for the zlib thing.

[image]

so, something seems to have went astray during then and now.

hope that helps

olliw42, 2019-10-29:
I now believe it has something to do with the "PairingManager" (whatever this is), and with commit 3918559bf, Merge pull request #7718 from mavlink/pairing. That's at least the commit where the INCLUDE and LIB paths were changed.

hope that helps too. :)

dogmaphobic, 2019-10-29:
Yup, that should have been added to a different code block (not related to SDL). The pairing code is still under development and normally not built by default.

olliw42, 2019-10-29:
there is really something screwed up

I just noted that the folder is named libs/zlib/windows/lib but in LIBS it is added as libs/zlib/windows/libs (note the last 's'), so I renamed the folder to libs, but it still fails to build with the very same error message ...

> normally not built by default.

which raises the question: why does it appear to be build for me, and how could I disable it being build?

dogmaphobic, 2019-10-29:
Because the declaration is in the wrong scope. In summary, Windows builds are broken until that gets fixed. I don't have a running Windows system to test that right now. I will see what I can do.

olliw42, 2019-10-29:
ah ... :)

thx at least for the confirmation that there seems to be an issue

is there any change which comes to your mind I might  test?

dogmaphobic, 2019-10-29:
Hopefully this solves it: https://github.com/mavlink/qgroundcontrol/pull/7959

olliw42, 2019-10-29:
YES, IT APPEARS TO SOLVE IT  !

what I did: I renamed the original QGCExternalLibs.pri, copied the QGCExternalLibs.pri of your PR into my folder. It compiles fine, and it also runs when I press the |>. 

THANKS!

pl allow me this question: 
Given that this exists since 23.Aug (or so), would I be mistaken if I would conclude that the Windows build is relatively rarely tested? How could there then had been (seemingly) working daily builds for Windows in the last two months?
I also ask because of https://github.com/mavlink/qgroundcontrol/issues/7869, and wonder now if that could also be a Windows specific thing

and also this one (although unrelated):
there are 15 of this deprecated str function warnings (strcpy which should become strcpy_s and so on)(the silencing with /wd4996 seems not to work). If I would change them to the safe functions would this break the non-Windows builds?

JonPompa, 2020-07-27:
Adding to a closed thread since it popped up again.  I cloned the master of QGC to a windows system and tried to run a make using the versions of Qt and MSVC as described on the QGC "gettting started" developer page. I ended up with a failed build with the "z.lib not found" error as well. It looks like zlib is back in the project for windows since this thread closed last year. 

output looks like:

> MANIFESTDEPENDENCY:type='win32' name='Microsoft.Windows.Common-Controls' version='6.0.0.0' publicKeyToken='6595b64144ccf1df' language='*' processorArchitecture='*'" /VERSION:0.0 /MANIFEST:embed /OUT:staging\QGroundControl.exe @C:\Users\JP\AppData\Local\Temp\QGroundControl.exe.10088.839312.jom
> LINK : fatal error LNK1181: cannot open input file 'z.lib'
> jom: C:\JP\Software\QGroundControl\build-qgroundcontrol-Desktop_Qt_5_12_6_MSVC2017_64bit-Release\Makefile [staging\QGroundControl.exe] Error 1181
> 14:34:05: The process "C:\Qt\Tools\QtCreator\bin\jom.exe" exited with code 2.
> Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.12.6 MSVC2017 64bit)
> When executing step "Make"

I'm hoping a hint to the issue is in the CMakeLists.txt file which says:
```
target_link_libraries(qgc
	PRIVATE
		shp
                Qt5::QuickControls2
                z # zlib
```

It seems like the private libraries group is where the build is referencing zlib as "z" instead of "zlib" and creating the broken reference. Does anyone know where the substitution is created?

DonLakeFlyer, 2020-07-27:
Don't use cmake and it will work. Looks like cmake is broken. cmake is not guaranteed to always work.

JonPompa, 2020-07-28:
Thanks DonLakeFlyer, I made sure it was running qmake not cmake - which I guess makes the cmake files irrelevant? The error still appears.

DonLakeFlyer, 2020-07-28:
This should have been fixed on latest master. Are you sure you up to date. Also delete your entire build directory and build again if you had an old build in there.

JonPompa, 2020-07-28:
All Fixed. Thank you. I brought up another machine with a fresh install of all the tools and it built on the first try. Sorry for the distraction.

lukegluke, 2020-07-31:
I also have linker issue: LNK1104: cannot open input file 'z.lib'

Windows 10, Qt 5.12.6, MSVC 15.9.28307.1216 x86_arm64
QGC 93205655 with #8955 fix (I wonder how one can reach the linker - without this patch build fails firstly on zlib.h not found?)

mnumanuyar, 2020-08-04:
for me, 4.0.9 and 4.0.8 builds fine, but master gives this error

DonLakeFlyer, 2020-08-04:
USe absolute latest master

rakesilverrian, 2020-08-10:
[image]
i get this on the latest build. How can i fixed it?

mmstewart14, 2020-08-19:
Here is what solved this issue for me:

If the _right version_ of [GStreamer](https://gstreamer.freedesktop.org) is installed on your machine, QGC will build successfully because it finds z.lib and some other required INCLUDE files it needs from GStreamer.

**Here's the Catch**: QGC will not successfully build with the newest GStreamer version installed on your machine. It seems you need an older version of the _MSVC_ _development_ installer. I only got my QGC to build by copying a colleague's older GStreamer program files into my C: drive.

DonLakeFlyer, 2020-08-19:
Ah, that's a hint I needed to figure this out. Working on it.

DonLakeFlyer, 2020-08-19:
> It seems you need an older version of the _MSVC_ _development_ installer

Requirements are here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md. Including links for where to get the right gstreamer.

luohao123, 2021-07-24:
can anyone tell how to install zlib on windows?