mavlink/qgroundcontrol #7955: Default text size too huge on recent daily builds

Antiheavy, 2019-10-28:
the default text size is defaulting to 20 on recent daily builds.  Is this intentional?  looks bad on all the laptops I've tried it on (Windows).

[image]
[image]

dogmaphobic, 2019-10-28:
QGC doesn't have a "default" font size. The size is computed at runtime the first time you run. From then on, it's a "setting", which you can adjust (and it will persist). Try the _Clear all settings on next start_ option and see if it comes back as 20. If it does, something is wonky with your system, making QGC compute the font size wrongly.

For reference, here is where the base font size is computed:
https://github.com/mavlink/qgroundcontrol/blob/master/src/QmlControls/ScreenTools.qml#L160-L196

The "default" set in `QGroundControl.settingsManager.appSettings.appFontPointSize` is 0. That causes the code within the condition to run (and compute the font size).

Antiheavy, 2019-10-28:
> Try the _Clear all settings on next start_ option and see if it comes back as 20. If it does, something is wonky with your system, making QGC compute the font size wrongly.

yep, it comes back as 20 pt font again.  This does not happen on QGC stable release though....  Is this auto-calculating stuff newer?  I'll play around and see if I can get it to behave correctly on my machine (Lenove window's laptop) and compare to other machines.

dogmaphobic, 2019-10-28:
No, for non mobile builds (as it is the case with Windows), the base font size is based off measuring an "X" character in pixels. That code has been the same for years.

DonLakeFlyer, 2019-11-18:
Something strange going on with that Windows setting to make fonts seem bigged on high density screens? @Antiheavy What are you display settings setup to?

DonLakeFlyer, 2019-11-18:
Only difference I see in ScreenTools on Stable is a different font:
```
Donalds-Mac-Pro:Downloads Don$ diff ScreenTools.qml ~/repos/qgroundcontrol/src/QmlControls/ScreenTools.qml 
48a49
>     property real availableHeight:          0
78c79
<     property bool isShortScreen:                    ((Screen.height / realPixelDensity) < 120) || (ScreenToolsController.isMobile && ((Screen.height / Screen.width) < 0.6))
---
>     property bool isShortScreen:                    ScreenToolsController.isMobile && ((Screen.height / Screen.width) < 0.6) // Nexus 7 for example
93d93
<     property real comboBoxPadding:                  defaultFontPixelWidth
95,96c95
<     // It's not possible to centralize an even number of pixels, checkBoxIndicatorSize should be an odd number to allow centralization
<     property real checkBoxIndicatorSize:            2 * Math.floor(defaultFontPixelHeight * (isMobile ? 1.5 : 1.0) / 2) + 1
---
>     property real checkBoxIndicatorSize:            Math.round(defaultFontPixelHeight * (isMobile ? 1.5 : 1.0))
99,100c98,99
<     readonly property string normalFontFamily:      ScreenToolsController.normalFontFamily
<     readonly property string demiboldFontFamily:    ScreenToolsController.boldFontFamily
---
>     readonly property string normalFontFamily:      "opensans"
>     readonly property string demiboldFontFamily:    "opensans-demibold"
Donalds-Mac-Pro:Downloads Don$ 
```

DonLakeFlyer, 2019-11-18:
> What are you display settings setup to?

This screen:
[image]

Antiheavy, 2019-11-18:
I think this "magically" stopped being a problem recently.  It must have been something weird with my machine(s)....  Today's daily build looks totally normal and defaults to a reasonable font size.  I'll close this.

DonLakeFlyer, 2019-11-19:
Hmm, that's not very satisfying :)

Antiheavy, 2019-11-21:
gah, the problem is back!  I'll try digging into this some more again.

Here is what my Laptop display settings are in Window's:
[image]

DonLakeFlyer, 2019-11-21:
Most likely caused by the 250% scaling for high dpi screen. Now the question how to work around that.

DonLakeFlyer, 2020-02-20:
@Antiheavy Is still still happening?

Antiheavy, 2020-02-21:
I think it must have something to do with multiple monitors....  if I completely uninstall QGC, reboot windows, have no external monitors connected, fresh install QGC (after uninstalling), then the UI scaling seems correct.  I think i've heard that QT doesn't play nice with multiple monitors so let's chalk it up to that.  I'll close this.

closed 2020-02-21 by Antiheavy