mavlink/qgroundcontrol #7956: Plan Wizard: suggested text for Takeoff dialog

Antiheavy, 2019-10-28:
After showing the [WIP] Takeoff dialog to a number of people here is some suggested improvements to the text:

Suggested:
"Move the P near the planned launch location and move the T to the desired climbout location."
"Move P (Planned Home) near the launch location and move T (Takeoff) to the climbout location."

Current:

[image]

tilaktilak, 2019-10-29:
We can probably have a "click-to-add' instead of making it appears on the center of the map and explain to the user that he has to move it ? That will make user experience more sensible.    

Also I think the Takeoff / Planned Home behavior is different depending on MultiCopter, Plane, ArduPilot, PX4, ... I put that here.

DonLakeFlyer, 2019-11-17:
> We can probably have a "click-to-add'

Done.

> here is some suggested improvements to the text:

I lost track of this one when I was making changes in #8049. I'll fix this up.

DonLakeFlyer, 2019-11-18:
Wording changes are done

closed 2019-11-18 by DonLakeFlyer