mavlink/qgroundcontrol #7958: Fixed wing landing pattern: glide slope altitudes too precise and cluttered

Antiheavy, 2019-10-29:
The altitude indications on the fixed wing landing pattern glide slope are both too precise and cluttered.

- The altitude indication at the loiter is a good idea so people can easily sanity check they are well above trees/power-lines/etc.
- I suggest getting rid of the mid-point and near-ground altitude indicators:  these dangerously imply that the vehicle can be expected to clear objects close to those heights.  When in fact, most times the fixed wing will be wandering significantly off the intended glide slope (depending on wind, baro error, delayed control, and other effects). 
- Showing a decimal (tenths of meters/feet) indicates too much precision that the vehicle would never be able to achieve anyway, plus it clutters up the screen.  Should we just round to the nearest meter/foot?

The biggest danger with the image below is that if there is a fence or short tree line at the start of the landing area, a user could claim the plan said it would clear the ground by 5.3 meters, when in fact it very likely could fly right into the fence or short trees.
[image]

ryanjAA, 2019-11-10:
I don’t know how you inform the user that the slope elevations are calculated numbers and not precise exactly as @Antiheavy mentioned but interestingly, I’m of the opposite mindset with regard to keeping them. I like that they are there. I’m fully aware of their lack of true precision in reality but they are ideal to give a sense of where things are on slope and during landing. We actually have basic trig app all our people have to use to quickly calc heights needed for clearance for prob 80% of flights. Maybe if people don’t like this make it an option in settings? Cool feature though.

DonLakeFlyer, 2019-11-11:
I think if you only show a single height indicator for the loiter point it is way less clear as to what exactly the little indicator is. With multiples going down the glide slope I think it makes it more obvious as to what these are.

If get the you have to be careful since they are really just an approximation. But I'm kind of with @ryanjAA as far as that approximation being useful for at least eyeballing the validity of the path.

How about this:
* Round up the value to integers to make the approximation even higher in altitude
* Add warning in editor talking about being careful with slope approximations.
* Maybe an "*" after each height value which then links to the "* Don't be an idiot with these values warning" in the editor.

DonLakeFlyer, 2019-11-11:
Corrections:
* The Loiter altitude indicator should match the precision and display of the Loiter Point altitude field in the editor. There should be no rounding/truncation on that on since it should be somewhat exact.
* Values should be rounded down not up for better safety.

Possible warning:
"* Intermeditate glide slope altitudes displayed are approximations only. Actual flight path can varied widely in height from these due to evironmental conditions and/or vehicle settings.

Antiheavy, 2019-11-11:
> Round up the value to integers to make the approximation even higher in altitude

Round down would be safer.

Antiheavy, 2019-11-11:
> Possible warning:
> "* Intermediate glide slope altitudes displayed are approximations only. Actual flight path can varied widely in height from these due to evironmental conditions and/or vehicle settings.

how about a shorter version: "Glide slope altitudes are approximate. Actual flight path will vary due to environmental conditions and vehicle settings."

Antiheavy, 2019-11-12:
https://github.com/mavlink/qgroundcontrol/pull/8015 Seems to satisfy the consensus.

closed 2019-11-12 by Antiheavy