mavlink/qgroundcontrol #7960: Use MAV_FRAME_GLOBAL when sending fences

rmackay9, 2019-10-30:
While testing complex fence uploads to ArduPilot from QGC I've found that they are being rejected (by AP) because the MAV_FRAME = 3 (MAV_FRAME_GLOBAL_RELATIVE_ALT) instead of the expected "0" (MAV_FRAME_GLOBAL).

We can modify AP to accept "3" but I think the frame being sent has changed since earlier testing so I'd like to be sure this is an intentional change.  Also considering there is no altitude sent for fences does it make sense to say it is relative-alt?

DonLakeFlyer, 2019-10-30:
I checked PX4 firmware and it's fine with both MAV_FRAME_GLOBAL AND MAV_FRAME_GLOBAL_RELATIVE_ALT.

DonLakeFlyer, 2019-10-30:
@rmackay9 So I hadn't bothered to look at the QGC code when we first discussed this. The QGC code has always been sending MAV_FRAME_GLOBAL.

Polygons: https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/GeoFenceManager.cc#L75
Circles: https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/GeoFenceManager.cc#L93

Rally points a relative though. Current UI doesn't support anything other than that.

rmackay9, 2019-10-30:
@DonLakeFlyer, very strange.  When I tested a few days ago using the latest daily build it seemed QGC was sending the frame as MAV_FRAME_GLOBAL_RELATIVE_ALT.

rmackay9, 2019-10-30:
.. in any case, I've relaxed the checks on the AP side..

DonLakeFlyer, 2019-11-08:
Maybe using the wrong build? Don't know. Anyway from what I can tell things seem correct so closing.

closed 2019-11-08 by DonLakeFlyer