mavlink/qgroundcontrol #7962: Advanced Tuning Setup Crashes QGC

dogmaphobic, 2019-10-30:
Adding here to keep track.
Hitting the _Advanced_ checkboxe causes QGC to segfault. It's somewhere deep within the QtChart module.

[image]

[image]

stefandunca, 2019-11-05:
Fix here: https://github.com/mavlink/qgroundcontrol/pull/7978

DonLakeFlyer, 2019-11-14:
Reopening since I backed out the QGuiApplication->QApplication change since it caused other bad problems.

DonLakeFlyer, 2019-11-14:
Also advanced tuning is removed from the build until a workaround is figure out

DonLakeFlyer, 2019-11-14:
@dogmaphobic Is tried doing what the Qml Charts docs say to switch QGuiApplication->QApplication in a small test app and it worked. Didn't you say this didn't work for you?

dogmaphobic, 2019-11-15:
> Didn't you say this didn't work for you?

If solves the QtChart problem but causes MessageBox to close the entire application on exit.

DonLakeFlyer, 2019-11-15:
But I though you made a small test app and were able to reproduce the MessageBox problem. I can't reproduce the MessageBox problem in a small test app. Hence it look like it has someting specific to do with QGC. Which I think it's means there wmay be some way to make QApplicaiton work.

dogmaphobic, 2019-11-15:
Weird... I just loaded the small test project and indeed, it all works. I could swear it was closing the app when I clicked the OK button within the MessageBox. This needs some digging...

DonLakeFlyer, 2019-12-11:
@dogmaphobic Fixed now?

closed 2019-12-11 by DonLakeFlyer