mavlink/qgroundcontrol #7966: QGC doesn't create camera GUI

L4ncelot, 2019-10-30:
Hi,

I'm using the latest master version of QGC on Ubuntu 16.04. I have created `MAVLink` interfaces for my custom autopilot and camera (Viewpro Z30TM). My autopilot redirects messages from/to the `MAVLink` camera interface I created. Autopilot itself works fine, but I have a little problem with the camera right now.

I've implemented the camera interface according to camera communication protocol described [here](https://mavlink.io/en/services/camera.html) and its parameter handling using `camera definition` file with `parameter protocol` described [here](https://mavlink.io/en/services/camera_def.html).

My initial problem was that the camera section of the vehicle setup didn't show up in `QGC`. I've read somewhere that I must specifically send `TRIG_MODE` to the `QGC` in order to activate it, which I did and it worked.

According to the camera communication protocol `QGC` should have created GUI with parameters specified in the camera definition file which is sent to the `QGC` through `HTTP`.

I did all of that and `QGC` is actually parsing the definition file (or it seems like anyway). But the GUI with all of those parameters doesn't get created at all. Here's a log from `qtcreator`:
```
CameraManagerLog: Hearbeat from  100
CameraManagerLog: _requestCameraInfo( 100 )
CameraManagerLog: _handleCameraInfo: Z30TM viewpro Comp ID: 100
CameraControlLog: Using cached camera definition file: "/home/l4ncelot/Documents/QGroundControl/Parameters/viewpro_Z30TM_001.xml"
CameraControlLog: Parsing camera definition
CameraControlLog: Current locale: "en_us"
CameraControlLog: New parameter: "ZOOM" Writable Readable
CameraControlLog: New parameter: "FOCUS" Writable Readable
CameraControlLog: New parameter: "DAY_MODE" Writable Readable
CameraControlLog: New parameter: "PIP_MODE" Writable Readable
CameraControlLog: New parameter: "FNG_TRCK_START" Writable Readable
CameraControlLog: New parameter: "FNG_TRCK_COO_X" Writable Readable
CameraControlLog: New parameter: "FNG_TRCK_COO_Y" Writable Readable
CameraControlLog: New parameter: "FNG_TRCK_CSIZE" Writable Readable
CameraControlLog: _initWhenReady()
CameraControlLog: _requestCameraSettings()
CameraControlLog: handleSettings() Mode: 1
CameraControlLog: handleSettings() Mode: 1
CameraControlLog: _requestStorageInfo()
CameraControlLog: handleStorageInfo: 40000 2 1 1 64000 24000
CameraControlLog: handleStorageInfo: 40000 2 1 1 64000 24000
CameraControlLog: _requestCaptureStatus()
CameraControlLog: handleCaptureStatus: 40000 0 0 0 0
CameraControlLog: Set Photo Status: QGCCameraControl::PhotoStatus(PHOTO_CAPTURE_IDLE)
CameraControlLog: handleCaptureStatus: 40000 0 0 0 0
```

And here's the definition file of the camera:
```
<?xml version="1.0" encoding="UTF-8" ?>
<mavlinkcamera>
    <definition version="1">
        <model>Z30TM</model>
        <vendor>Viewpro</vendor>
    </definition>
    <parameters>
        <parameter name="ZOOM" type="uint32" default="0">
            <description>Zoom</description>
            <options>
                <option name="stop" value="0"/>
                <option name="zoom in" value="1"/>
                <option name="zoom out" value="2"/>
            </options>
        </parameter>
        <parameter name="FOCUS" type="uint32" default="0">
            <description>Focus</description>
            <options>
                <option name="stop" value="0"/>
                <option name="focus in" value="1"/>
                <option name="focus out" value="2"/>
            </options>
        </parameter>
        <parameter name="DAY_MODE" type="uint32" default="0">
            <description>Day mode</description>
            <options>
                <option name="day" value="0"/>
                <option name="night" value="1"/>
                <option name="auto" value="2"/>
            </options>
        </parameter>
        <parameter name="PIP_MODE" type="uint32" default="0">
            <description>Picture in picture mode</description>
            <options>
                <option name="visible and infrared" value="0"/>
                <option name="infrared only" value="1"/>
                <option name="infrared and visible" value="2"/>
                <option name="visible only" value="3"/>
            </options>
        </parameter>
        <parameter name="FNG_TRCK_START" type="uint32" default="0">
            <description>Finger track start/stop</description>
            <options>
                <option name="stop" value="0"/>
                <option name="start" value="1"/>
            </options>
        </parameter>
        <parameter name="FNG_TRCK_COO_X" type="uint32" default="0">
            <description>Finger track x coordinate</description>
        </parameter>
        <parameter name="FNG_TRCK_COO_Y" type="uint32" default="0">
            <description>Finger track y coordinate</description>
        </parameter>
        <parameter name="FNG_TRCK_CSIZE" type="uint32" default="0">
            <description>Finger track cursor size</description>
            <options>
                <option name="small" value="0"/>
                <option name="medium" value="1"/>
                <option name="big" value="2"/>
                <option name="small_medium_auto" value="3"/>
                <option name="all_auto" value="4"/>
                <option name="medium_big_auto" value="5"/>
            </options>
        </parameter>
    </parameters>
</mavlinkcamera>
```

Is there something else I need to specify for `QGC`  to create the GUI? Did I miss something crucial? I cannot find any other information related to this problem and I'm in kind of desperate state right now :D.

Thank you in advance for any advice.

dogmaphobic, 2019-10-30:
>My initial problem was that the camera section of the vehicle setup didn't show up in QGC. I've read somewhere that I must specifically send TRIG_MODE to the QGC in order to activate it, which I did and it worked.

That whole thing is for "dumb" cameras where all you can do is to trigger the shutter through a PWM (RC) signal. If you implemented the camera API, you don't need any of that.

The UI is likely there. It's just not as intuitive to find it as it should be. The camera control in the regular QGC is quite bare bones. Coincidentally, it's getting some long needed TLC just today (see https://github.com/mavlink/qgroundcontrol/pull/7963).

The camera "widget" is within that everythingunderthesunbutthekitchensink dialog:

[image]

Once there, you click in the (poorly colored) gear icon to access the settings (the one built off your XML definition file):

[image]

Are you building a custom version of QGC or using the stock version off master? Those tracking parameters you have seem to only make sense if you add proper UI to handle it (as in touching the screen and updating those parameters as you move around to define the tracking target).

Also, there are direct commands for zoom and focus.... which seems to be missing from the documentation (@hamishwillee? We missed these) If you [define basic support for zoom and/or focus](https://mavlink.io/en/messages/common.html#CAMERA_CAP_FLAGS), QGC will (in theory) put up proper zoom and focus UI for you. You would then implement the [ZOOM](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_ZOOM) and [FOCUS](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_FOCUS) commands within your camera handler.

L4ncelot, 2019-10-31:
@dogmaphobic I'm using the stock version of master right now. I cannot thank you enough for the info. 
 
I didn't know you can control the camera like this. I was thinking the whole time that the GUI should be in the "dump" camera settings. Sometimes the documentation can be a little bit misleading. When I had a look at the section you pointed it was right there and it works.

Also, thank you for providing me with the information about zoom and focus, it'll be pretty handy when I implement it.

closed 2019-10-31 by L4ncelot

gChuNguyen, 2020-05-28:
Sorry for bother you. 

Currently, we want to use the GUI from definition file to generate the UI for our camera and gimbal. But I have got stuck when using the camera manager to test first. I have got the CameraManagerLog: Hearbeat from  100
CameraManagerLog: _requestCameraInfo( 100 )
CameraManagerLog: Hearbeat from  101
CameraManagerLog: _requestCameraInfo( 101 )
CameraManagerLog: _requestCameraInfo( 100 )
CameraManagerLog: _requestCameraInfo( 101 )

Please show me where the data will go. TELE1/2 or somewhere.

DonLakeFlyer, 2020-05-28:
Best place to ask this sort of thing would be on the QGC channel in PX4 Slack.