mavlink/qgroundcontrol #7980: MacOS: crashes with bluetooth joystick connection

jaxxzer, 2019-11-06:
I've seen several reports over some period of time about crashes on MacOS when a joystick is connected by bluetooth.

Mac OS X 10.13.5 (17F77):
[qground 2.txt](https://github.com/mavlink/qgroundcontrol/files/3815467/qground.2.txt)

Mac OS X 10.15.1 (19B88):
[qgc_crash_mac.txt](https://github.com/mavlink/qgroundcontrol/files/3815468/qgc_crash_mac.txt)

DonLakeFlyer, 2019-11-08:
Looks like pretty generic crashes in the SDL framework. Could one of your folks look to see if we are using the latest version of the SDL framework. There may be a fix for it in newer versions.

jaxxzer, 2019-11-11:
@patrickelectric @Williangalvani @tcanabrava fyi

patrickelectric, 2019-11-11:
I have two logs here:
[qgc_crash.txt](https://github.com/mavlink/qgroundcontrol/files/3832029/qgc_crash.txt)
[qgc_crash2.txt](https://github.com/mavlink/qgroundcontrol/files/3832030/qgc_crash_mac.txt)

DonLakeFlyer, 2019-12-11:
@patrickelectric Can you get this to crash in the debugger as opposed to a crash log and do some poking around as to why. I can't repro so stack trace doesn't help much.

patrickelectric, 2019-12-17:
Hi @DonLakeFlyer, sorry for not saying this before, but it's in my todo list.

patrickelectric, 2020-01-15:
I'm doing some tests here but for now unable to replicate it, will try more and see how it goes

patrickelectric, 2020-01-15:
I'm unable to replicate it with the last version of the master branch [sdl2 2.0.10] and macOS [10.15.2]

patrickelectric, 2020-01-16:
Still unable to replicate, will close it for now.

closed 2020-02-11 by DonLakeFlyer