mavlink/qgroundcontrol #7981: Issues with firmware update on Windows

jaxxzer, 2019-11-06:
I've encountered this problem on a fresh install of windows 10 and QGC 3.5.5.

I'm experiencing two different issues appear during firmware update:
1. The bootloader exits, the autopilot program starts, qgc connects and the firmware update process exits: [FirmwareUpdaterLog](https://github.com/mavlink/qgroundcontrol/files/3816448/fmuv2BLskip.txt)
[image]

2. The bootloader apparently hangs (led blink stops, autopilot program never starts), and qgc firmware update process stalls: [FirmwareUpdaterLog](https://github.com/mavlink/qgroundcontrol/files/3816450/fmuv2BLhang.txt)
[image]

In testing, I'm using pixhawk mini (bootloader v4), two pixhawk1 (bootloader v4) from different manufacturers, and pixhawk2 (booloader v5).

The pixhawk mini is connecting and doing the firmware update perfectly every time. The others are having problems regularly and repeatedly, but indeterminately. Sometimes the issue is 1, sometimes the issue is 2, and (less often) it works. The results were the same with/without usb hubs or on usb2/usb3. Because these boards are all fmuv2, I don't know what to think about the pixhawk mini acting differently.

jaxxzer, 2019-11-06:
Possibly related to https://github.com/mavlink/qgroundcontrol/issues/7656

DonLakeFlyer, 2019-11-08:
This all seems like hardware problems. For 1, QGC just can't open the port to the board. For 2, it looks like the open succeeded but then QGC is likely getting stuck writing to the port. Does QGC hang for the second case? I did a code inspection for that path and it should be able to handle things going wrong on the port in that case (still failing firmware flash though). But I don't see any further messages in the log which indicate that.

DonLakeFlyer, 2019-12-11:
Closing since all seems like hardware/firmware side. If you can verify it's QGC side problem on latest master I'll take look again. At this point 3.5 isn't relevant since new Stable is about to go out.

closed 2019-12-11 by DonLakeFlyer