mavlink/qgroundcontrol #8003: mainWindow.showMessageDialog click Ok, entire app close

DonLakeFlyer, 2019-11-09:
Repro steps:
* Go to Plan view
* Click Survey in the Create Plan thingy
* File/Save As
* Dialog pops up about incomplete data
* Click Ok
* App closes

dogmaphobic, 2019-11-09:
I don't have the answer yet but I've located the source. It started with this fix: https://github.com/mavlink/qgroundcontrol/pull/7978

`ChartView` in PIDTuning.qml was crashing QGC. @stefandunca found this:
https://bugreports.qt.io/browse/QTBUG-49895

That fixed the `ChartView` crash but now it seems `MessageDialog` is causing issues if `QApplication` is used instead of `QGuiApplication`.

dogmaphobic, 2019-11-09:
It must be some weird combination of Qt modules or something. Putting together a bare minimum QML app showing both a `ChartView` and a `MesageDialog` works just fine (only if using `QApplication`):

[image]

If you do the same in QGC, the entire applications closes when you dismiss any `MesageDialog`. Just go to General Settings, check _Clear all settings on next start_ and hit "No". It closes all of QGC when using `QApplication` (it works fine when using `QGuiApplication` but then `ChartView` crashes.)

The bare minimum app:

```javascript
import QtQuick          2.12
import QtQuick.Window   2.12
import QtQuick.Controls 2.4
import QtQuick.Dialogs  1.3

import QtQuick          2.0
import QtCharts         2.0

Window {
    visible:            true
    width:              1280
    height:             800
    title:              qsTr("Hello World")

    Rectangle {
        anchors.fill:   parent
        color:          "black"
    }

    MessageDialog {
        id:                 simpleMessageDialog
        title:              "Test Dialog"
        text:               "Test Text"
        standardButtons:    StandardButton.Ok
        modality:           Qt.ApplicationModal
        visible:            false
    }

    ChartView {
        anchors.centerIn:   parent
        width:              400
        height:             300
        theme:              ChartView.ChartThemeBrownSand
        antialiasing:       true
        PieSeries {
            id: pieSeries
            PieSlice { label: "eaten"; value: 94.9 }
            PieSlice { label: "not yet eaten"; value: 5.1 }
        }
    }

    Button {
        text:                   "Click Me"
        anchors.bottom:         parent.bottom
        anchors.bottomMargin:   20
        anchors.horizontalCenter: parent.horizontalCenter
        onClicked:              simpleMessageDialog.open()
    }
}
```

DonLakeFlyer, 2019-11-10:
Sigh. More fun with Qt.

DonLakeFlyer, 2019-11-10:
FYI: I backed out the change which caused this for now.

feemi, 2019-11-11:
i was struggling with this issue also but couldnt find a way to debug. Is there a way to trace it?

DonLakeFlyer, 2019-12-11:
@dogmaphobic This is all good now right?

stefandunca, 2019-12-11:
@DonLakeFlyer it is fixed by https://github.com/mavlink/qgroundcontrol/pull/8074

closed 2019-12-11 by stefandunca