mavlink/qgroundcontrol #8011: Plan toolbar: Azimuth/Heading show same value

DonLakeFlyer, 2019-11-11:
Azimuth shoudl be change in heading

hamishwillee, 2019-11-11:
Thanks. I understand heading is relative to north. Are you saying azimuth is relative to the line connecting the previous waypoint?

DonLakeFlyer, 2019-11-12:
> Are you saying azimuth is relative to the line connecting the previous waypoint?

Correct

DonLakeFlyer, 2019-11-12:
So this is working correctly. It just confusing:
* Azimuth - Angle from previous waypoint to current
* Heading - Angle from current waypoint to next

hamishwillee, 2019-11-12:
@DonLakeFlyer Based on QGC 3.5.5 and your definition I still think this is "a little" broken. If the heading is angle is from current waypoint to next then surely it should be NA if you're creating a new waypoint and there is no next waypoint?

DonLakeFlyer, 2019-11-14:
The heading value is a bit odd. It is the expected vehicle heading at the waypoint. If there is no next waypoint then the heading just keeps going in the current direction. If there is another waypoint, then the heading points to that next waypoint.

hamishwillee, 2019-11-14:
OK, so:
- Azimuth - Heading from previous waypoint to current waypoint.
- Heading - Vehicle heading at waypoint (to next waypoint). For the final waypoint this is the expected heading (ie the path of the line to the current waypoint).

If that's reasonable docs this can be closed (and I'll update docs).

DonLakeFlyer, 2019-11-14:
Correct. And all of which is so not memorable it makes me wonder why it's in there in the first place.

closed 2019-11-14 by DonLakeFlyer