mavlink/qgroundcontrol #8016: 'mavlink_types.h' file not found

devd77, 2019-11-11:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

closed 2019-11-12 by DonLakeFlyer

devd77, 2019-11-12:
--I have cloned this repository on my system by "git clone https://github.com/mavlink/qgroundcontrol.git --recursive"
--and then run command: git submodule update
--then it on QT Creator
--Build for iOS Simulator Debug
--Then i have found "mavlink_types.h' file not found" this issue.

It was working before, but in new clone i have found this issue
Anyone can tell me the solution
Thanks in Advance :)

tilaktilak, 2019-11-12:
Not sure it's your problem but you can try ```git submodule update --init --recursive``` instead of ```git submodule update```

devd77, 2019-11-12:
> git submodule update --init --recursive

Thanks, but its still showing same error

devd77, 2020-07-11:
Same problem, not finding any solution

DonLakeFlyer, 2020-07-11:
Delete and start over from scratch again

devd77, 2020-07-11:
> Delete and start over from scratch again

I have tried but facing same problem
i am using QT 5.12.5 to build for iOS

DonLakeFlyer, 2020-07-11:
Qt would have nothing to do with it although you should be using 5.12.6 which is the supported version. This is just git problems unrelated to QGC.

devd77, 2020-07-11:
> Qt would have nothing to do with it although you should be using 5.12.6 which is the supported version. This is just git problems unrelated to QGC.

Thanks, will try from scratch again