mavlink/qgroundcontrol #8021: ComboBoxes not sizing correctly

DonLakeFlyer, 2019-11-12:
[image]

olliw42, 2019-11-12:
IMHO they not only sometimes don't size properly, but also look ugly in general!

I don't like the double arrows
I don't like the arrows to have such a large gap from the right edge
I don't like the arrows to be that big
I don't like that the text has such a small gap from the left edge
I don't like its outline is so high, so much higher than a button or text field
I don't like that its dropdown is so spaceous, which such large spacing between the text
they just don't fit well into the overall design but stick out as odd
you get it, I don't like the new combo boxes, and would very much appreciate if some if not all of these points would be improved
but I also understand that design is highly subjective, so just my 2 cents :)

DonLakeFlyer, 2019-11-12:
I agree with most of that. They were changed a while back and they are somewhat nasty looking.

Antiheavy, 2019-11-12:
> somewhat nasty looking

/agree

DonLakeFlyer, 2019-11-14:
I think this is good for now. I'll may come back to it.

closed 2019-11-14 by DonLakeFlyer

olliw42, 2019-11-14:
it indeed has MUCH improved !! 
Many thx.