mavlink/qgroundcontrol #8023: new Insert functionality buggy and confusing with Plan Wizard

Antiheavy, 2019-11-13:


Antiheavy, 2019-11-13:
and dis-jointed missions can be created:
[image]

Antiheavy, 2019-11-13:
FYI @tilaktilak

tilaktilak, 2019-11-13:
I'll have a look at this. 
For the record, I'm pretty sure the dis-jointed thing was there before if you insert a pattern from the Hamburger menu, to be checked.

tilaktilak, 2019-11-13:
That's a display problem (as the mission item list is consistent) and it's the same problem in both cases actually : 
As you do not set the takeoff neither the Survey and "Return to Landing" is checked there is a disjunction in your flight plan. 
As soon as you define everything your WP will appear at the end of the flight plan. 

Not sure how to fix that. From my understanding, we can't really display something sensible since the Takeoff or the Survey is not defined. Should we block the user to insert anything until he completes every step?

 @DonLakeFlyer what do you think of it?

DonLakeFlyer, 2019-11-13:
Let me work through what I'm doing with a contextual left toolbar and then I'll come back to this. Likely that incomplete pattern items need to provide a temporary coordinate until the polygon/line is completed.

DonLakeFlyer, 2019-11-13:
This doesn't have anything really to do with the new insert. It's a quirk/bug of the new Pattern stuff.

Antiheavy, 2019-12-17:
Random idea:
How about drawing a dashed red line between the dis-jointed two waypoints with a '?' icon similar to the red icon on the incomplete mission item?  Something like this:
[image]

PR #8594 (2020-03-24): Plan: Remove display of incomplete item line connections
* This essentially backs out the change to fix #8023. The correct fix for this is way more complex than I originally thought. And the existing code was buggy and causing crashes. Not possible to deal with in current stable. Will have to wait for point release.
* Fixed item sequence numbering problem for incomplete items. This could cause more than one item to have the same sequence number. Which would in turn cause things like multiple items being marked as current at the same time.