mavlink/qgroundcontrol #8045: Latest daily build closes after connection

highfreq, 2019-11-15:
Tried latest build today, it closes itself after 10-20 seconds.

Everythong works, it connects to vehicle and starts video but after a short while it closes with no errors.

I am still running 3.6.8, don’t know if it needs latest stable to work correctly.

It happens in compatibility and safe mode too.

It happens only if the uav is connected to the ground station, if i don’t connect it than QGroundcontrol never closes. As soon as i turn on the uav and it connects and finishes fetching the parameters than QGC closes.
I am running it on windows 10 64bit pc.

DonLakeFlyer, 2019-11-15:
Instructions here: https://docs.qgroundcontrol.com/en/Support/Support.html will show you where to get the crash dump file for QGC. Zip up that file, put it someplace and link to it here.

highfreq, 2019-11-15:
dump:

https://mshelisrl-my.sharepoint.com/:u:/g/personal/davide_monti_mshelisrl_onmicrosoft_com/EenVINRfYCNIg-gFbUdZW7kBKfKD7BBizqM7E1rW4vMTJA?e=Plo5dy

DonLakeFlyer, 2019-11-16:
I can't seem to get that dump file to load correctly. Possible because the daily build has moved past the version you are using. Can you instead install this special daily build from here: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl-installer.exe. Get it to crash again and link to the new dump file.

highfreq, 2019-11-16:
Here is the dump from the linked version. This version doesn't even start.

https://mshelisrl-my.sharepoint.com/:u:/g/personal/davide_monti_mshelisrl_onmicrosoft_com/EauqwOFdvs9Hhop9Q2TeQkIBvwJJXIvWsWfKB-PUWV7fKg?e=I0eiAz

DonLakeFlyer, 2019-11-16:
Sorry my screw up. I was testing dmp file debugging with QGC and put in a forced crash for testing. Forgot to take it our before I built the installer! There is a new installer which doesn't crash (at least right away) at the same link.

highfreq, 2019-11-17:
The one you linked doesn't crash, but i can't find where to use additional joystick axis for the gimbal and video looks like is not configurable.
Is it a special version to just find the crash problemù?

DonLakeFlyer, 2019-11-17:
Should be fixed now in latest released daily builds. Could you verify?

highfreq, 2019-11-18:
Ok thanks now it works.
It is sad it is useless for me because gimbal has no joystick assignment and no mount control on joystick. I would really love to use it but as it is now i have no gimbal control. Thank you anyway.

closed 2019-11-19 by DonLakeFlyer