mavlink/qgroundcontrol #8056: App menus missing in Win 10

mwiatt, 2019-11-19:
## Expected Behavior
App menus should be available in Win 10.

https://docs.qgroundcontrol.com/en/app_menu/

## Current Behavior
Menu bar is not present.

## System Information
When posting bug reports, include the following information
- Operating System: Win 10
- QGC Version: Daily 11/18/2019

## Detailed Description
I looked in the old issues and PRs to find where this might have been removed but didn't come up with anything.

## Log Files and Screenshots
[image]

DonLakeFlyer, 2019-11-19:
That's because there is longer a menu in any QGC build. The menu had only a few items in it if I remember:
* Show replay status bar. This is now available in Settings/General.
* Show Mavlink Inspector. This is now available from the Analyze view. Last toolbar button which is a piece of paper with a magnifying glass.
* Charting. This is currently unavailable in daily bulds but should hopefully come back soon. Likely also will come from the Analyze view. But TBD.

closed 2019-11-19 by DonLakeFlyer

mwiatt, 2019-11-19:
There are a few other things missing that I used.

1) The file transfer tool. This is useful for changing mixer files on the SD card.
2) The custom command widget. I use this for things like gimbal testing.

DonLakeFlyer, 2019-11-19:
The file transfer tool feature was removed a while back, I think even before current Stable. Mainly because it was buggy as hell. It only worked over USB and even then had a high failure rate.

Not sure if the custom command widget will come back. It theoretically should but isn't currently listed as one of the things to put back. What do you need for gimbal testing?

mwiatt, 2019-11-19:
It was just an easy way to issue ROI and DO_MOUNT commands while on the ground. I was doing this to test the entire path of the PX4 firmware. I am having some (a lot) of problems getting ROI commands to be processed while in flight. Not a big deal. I can use an old version of QGC when testing.