mavlink/qgroundcontrol #8069: Suggestion: more landing pattern tweaks

Antiheavy, 2019-11-24:
Here are a handful of suggested tweaks to the Landing Pattern:

- Consider even less wordy caution text for glide slope: “* Approximate glide slope altitudes.  Actual flight path will vary.”

- Use a orange/yellow text for the caution text instead of red.  Red should typically be reserved to indicate a problem state.

- Match the color of the glide slope altitude numbers to the caution text.  Feedback we've received from users is that these values should be an indication of caution, vs an indication of safety.

- Add new caution text above "Heading" that says: "Avoid tailwind on landing." or something similar.

[image]

DonLakeFlyer, 2019-12-25:
> * Consider even less wordy caution text for glide slope: “* Approximate glide slope altitudes.  Actual flight path will vary.”

Done

> * Match the color of the glide slope altitude numbers to the caution text.  Feedback we've received from users is that these values should be an indication of caution, vs an indication of safety.

This get pretty tricky when displaying text over a map. There is a special pallete use and outlining such that the text is visuals when show above all different map types as a background. I tried yellow and red text and it just doesn't work on all map types.

> * Use a orange/yellow text for the caution text instead of red.  Red should typically be reserved to indicate a problem state.

This is currently a palette color. In order to change I have to look at all other use to see of yellow or orange works. To close to stable release to deal with that.

DonLakeFlyer, 2019-12-25:
> * Add new caution text above "Heading" that says: "Avoid tailwind on landing."

Done

DonLakeFlyer, 2019-12-25:
This changes as much as I can for Stable

closed 2019-12-25 by DonLakeFlyer