mavlink/qgroundcontrol #8070: Plan Wizard: Takeoff text when no vehicle connected

Antiheavy, 2019-11-24:
When not connected to a vehicle, add additional wizard text to the Takeoff item to move the Launch location as well as the Takeoff location.  Suggested text:

**When not connected to a vehicle with a Home location:**
"Move L 'Launch' near the planned launch location."
"Move T 'Takeoff' to the climbout location."
"Ensure clear of obstacles and into the wind."

**When connected to a vehicle with a Home location:**
"Move T 'Takeoff' to the climbout location."
"Ensure clear of obstacles and into the wind."

[image]

DonLakeFlyer, 2019-12-11:
That is how it should already be working but I'll double check

DonLakeFlyer, 2019-12-11:
So this works differently now when not connected. The Launch position is set by a click, and the T is then moved if needed. So I think this is all good.

closed 2019-12-11 by DonLakeFlyer