mavlink/qgroundcontrol #8072: Sensors pane showing all sensors ready, with failed sensors

steffensol, 2019-11-26:
I do not know if this bug is caused by a combination of QGC or PX4's method of reporting sensor faults/statuses - however i would like the teams to look at this together. I will be posting a link to this bug report also on the PX4-FW github, and hope that this is acceptable. Hopefully we will get a good solution out of this.

Prequel;
Our facilities lie within the controlled airspace of a local airport, so arming a vehicle at this location is a no-go. When our pilots go into field, we would like to be able to verify that our drones are operational and ready to go to work when they get there. We've been basing this test on the "Sensors pane" under the QGC vehicle Summary. Typically we're seeing that the sensors and calibrations pages are currently not doing a good job at reflecting the actual status of the connected vehicle. Sometimes we're seeing vehicles getting out of calibration, and other times it is reflecting the wrong status of external/internal sensors.

Specific issue:
Running V3.5.4 - with PX4 1.9.2;
During preflight phase it seems that QGC is incorrectly showing all my sensors as "Ready" in the Sensors list, under the Summary pane - when they have failed.
[image]

## Expected Behavior
With vehicle in a bad state - e.g. with TOUT/ECNT/EDNST on Accel/Gyro sensors, i would expect QGC to show them as failed/inconsistent or with some other error in the Summary pane, so that i would know that my sensors have failed. 

## Current Behavior
With vehicle in a bad state - e.g. with TOUT/ECNT/EDNST on Accel/Gyro sensors, the sensors pane is showing "Ready" on all sensors - when they have actually failed. (See screenshot above)

However, when running a sensors check through the console - it clearly states that there are errors.

[image]

If i try to arm the vehicle at this point of time i get Accelerometer/Gyro failures and no arm will be allowed - which is as expected.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Get a FC that has sensor faults, or create one by disconnecting an external sensor, e.g. compass
2. Connect to FC through QGC and go to the vehicle summary.
3. QGC will show all sensors as Ready
4. Go to mavlink console and run ```sensors status``` - it should report errors.
5. Try to arm vehicle - and it should fail.

## System Information
- Operating System: Ubuntu 18
- QGC Version: 3.5.0
- QGC build: Stable
- Flight Controller: Drotek Pixhawk 3 Pro
- Autopilot (with version): PX4 1.9.2 - Stable - From PX4 buildsystem

DonLakeFlyer, 2019-11-26:
The set of pages you are referring to are "Setup" pages. They only show status as to whether the vehicle setup has been completed. They are not meant to reflect general vehicle status. So as far as what this QGC UI is meant to do today this is all working correctly.

steffensol, 2019-11-27:
Thanks for following this up @DonLakeFlyer - I truly appreciate that you are looking into this. I understand the initial thought about this tool showing if a calibration has been completed or not - and i think that part of the UX is quite good. 

I would still wager that it would be beneficial to many users if we could get additional information regarding the vehicle state through the Summary-tab. @dagar from the PX4 team mentions a specific MAVLink message containing at least some parts of the needed information (in the PX4 sidereport). Would it make sense for me to propose that this information should be included?

Our pilots are currently reporting that they have seen multiple times where we receive a calibration inconsistency error while still seeing this Ready status. I will at least inform them that the sensor statuses in that page does not give them the information that they are trying to find - and that they will have to run the sensors diagnostics/status in the console. This unfortunately only works for some of our vehicles, as console is disabled in one of our vehicle types.

DonLakeFlyer, 2019-12-11:
> I would still wager that it would be beneficial to many users if we could get additional information regarding the vehicle state through the Summary-tab.

Don't disagree. Requires a lot of new mavlink telemetry from the vehicle to implement though which does not currently exist. I'd propose starting a discussion on the mavlink repo (linking tho this issue) as to what is needed/missing so folks can start to propose new telemetry. Once that gets done then the firmware needs to implement and finally last in the chain is QGC support.

closed 2019-12-11 by DonLakeFlyer