mavlink/qgroundcontrol #8081: Feature Request: default survey transects perpendicular to takeoff direction for fixed wing

Antiheavy, 2019-12-02:
QGC remembers your many of your last used settings in the Survey complex mission item tool.  Last used transect direction, in particular, is somewhat arbitrary and likely isn't exactly what is wanted.  

For fixed wing aircraft we can have a better default based on assumed wind direction:
- If a takeoff mission item is already present in the mission, it can be assumed that the takeoff direction is generally into the wind.
- For most small fixed wing drones, it is preferable to start on the downwind side of the survey area and fly direct crosswind transects working your way upwind as you go.
- Can we make the initial default for the Survey pattern be perpendicular to the takeoff direction using the above information?
- If no takeoff item present, then default to the last used (or maybe default to the longest straight side of the survey area).

Example:
[image]

DonLakeFlyer, 2019-12-24:
Doable but not simple. Moving to 4.1

Antiheavy, 2020-04-03:
This feature could also be useful for many VTOLs and Multicopters.  Some multicopter users prefer to fly mapping missions cross-wind because the flight path results in more consistent pitch angles on the vehicle.  It also is more efficient for power consumption based on aerodynamic drag as power due to drag increases with airspeed-cubed.  Flying into a headwind uses a lot more power for the same ground speed.