mavlink/qgroundcontrol #8082: Feature Request: more distict graphics for Takeoff

Antiheavy, 2019-12-02:
A lot of the QGC mission plan graphics look similar and it can be hard to pick out certain important items.  Takeoff is one example of this.  It would be nice if Takeoff was somewhat more distinct than the normal mission and transect lines shown for mission items at altitude.

Here is a simple idea for a triple chevron graphic clearly showing the direction of takeoff and indicating the vehicle is climbing off the ground up to altitude (by the increasing size of the chevrons).  This is just an idea, I'd invite others to make suggestions too, the main goal is for Takeoff to be a bit more distinct relative to normal mission items.
[image]

hamishwillee, 2019-12-02:
@Antiheavy FMI, is "Launch" the new name for "Planned Home"?

Antiheavy, 2019-12-02:
There have been some changes here recently, I don't know how final things are or how they work with fixed wing vs multicopter and PX4 vs Arducopter.  @DonLakeFlyer would know more.  FWIW the fixed wing folks we work with like the terminology "Launch" much better than "Planned Home".

This could be a place to start if you want to see the history:
https://github.com/mavlink/qgroundcontrol/pull/8049

DonLakeFlyer, 2019-12-11:
> is "Launch" the new name for "Planned Home"?

Yes.

DonLakeFlyer, 2019-12-11:
> I don't know how final things are or how they work with fixed wing vs multicopter and PX4 vs Arducopter.

Mostly at bug fix stage now. There is all sorts of new handling with respect to Launch+Takeoff and MR or FW vehicles.