mavlink/qgroundcontrol #8085: Disarm indicator running off edge of 8" tablet screen with VTOL (Master)

dlwalter, 2019-12-03:
## Expected Behavior
Built Master running on 8" Samsung Android Tablet - all icons on top toolbar should be visible

## Current Behavior
With the addition of the Mavlink Inspector icon, the Disarm indicator/button is running off the edge of the screen

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Current Master build (03917bd8c)
2. Connect to VTOL Airframe

## System Information
When posting bug reports, include the following information
- Operating System: Android 7.0
- QGC Version: 3.5.x
- QGC build: self-build 03917bd8c
- Flight Controller: The Cube, Black
- Autopilot (with version):PX4 1.9.2

## Log Files and Screenshots
Stable 3.5.1 tag:
[image]

Current Master
[image]

## Proposed Fix
1. Change VTOL indicator from "VTOL: Multi-Rotor" to "VTOL:MR" and "VTOL:FW" or removed "VTOL" and just have the mode multi rotor vs fixed wing.
2. Put the GPS satellite count inside the sat icon and lose the HDOP number.  Operators can access this info by pressing on the satellite.

dlwalter, 2019-12-04:
I just noticed that in the current Daily Build the satellite info is not shown on the toolbar, only the icon.   I don't see this change on master (b771144af)

DonLakeFlyer, 2019-12-11:
Can you double check new version?

closed 2019-12-11 by DonLakeFlyer