mavlink/qgroundcontrol #8086: QGC using MISSION_ITEM rather than MISSION_ITEM_INT when uploading to AP

rmackay9, 2019-12-04:
I've noticed that when QGC is uploading missions to AP (i.e. Copter-4.0.0) it's using the less accurate MISSION_ITEM message rather than MISSION_ITEM_INT.  The equvalent report for ArduPilot is over here: https://github.com/ArduPilot/ardupilot/issues/12961

DonLakeFlyer, 2019-12-10:
QGC requires a response to the AUTOPILOT_VERSION message with the item_int capability bit set for it to use MISSION_ITEM_INT. I don't have an ArduPilot setup on my laptop right now (and don't have enough internet to set it up) so I can't test ArduPilot. But I did test PX4 and it worked as expected. There "shouldn't" be a difference in this negotiation between PX4 and ArduPilot.

If you turn on logging for VehicleLog it will show you the capabilities which are being set at first connect. You should also be able to see the AUTOPILOT_VERSION message which QGC saw in the Mavlink Inspector.

peterbarker, 2019-12-12:
@DonLakeFlyer I've just had another play with this.  I can confirm that it is using `MISSION_REQUEST_INT` to *download* items correctly.  However, at the same time, it is *uploading* items using `MISSION_ITEM`.  And the reason is... it's kinda working according to spec!

Both MissionPlanner and MAVProxy work out-of-spec; they convert requests for `MISSION_ITEM` into `MISSION_ITEM_INT` - and ArduPilot only ever sends requests for `MISSION_ITEM` (https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/MissionItemProtocol.cpp#L318).  MissionPlanner checks the vehicle's compatibility bits before doing that.

For PX4 Firmware (If my reading is correct), if you don't download the items first (giving the GCS an opportunity to indicate it can do _INT), it will MISSION_ITEM.  I haven't delved, but it looks like you may get interesting effects with the `_using_int` state if there are multiple GCSs floating around.

Really, there should probably have been `MISSION_COUNT_INT` and `MISSION_REQUEST_LIST_INT` to disambiguate all of this.

DonLakeFlyer, 2019-12-12:
@peterbarker ArduPilot doesn't send the mission item int capability bit? That is what triggers upload int usage.

DonLakeFlyer, 2019-12-12:
I do wonder though if not using MISSION_ITEM_INT should just be deprecated completely. And all GCS should just use that only and not bother with checking capability or whatever?

@hamishwillee FYI

peterbarker, 2019-12-12:
On Wed, 11 Dec 2019, Don Gagne wrote:

> @peterbarker ArduPilot doesn't send the mission item int capability bit? That is what triggers upload int usage.

It does.

But we *request* items from the GCS using MISSION_REQUEST, *not* 
MISSION_REQUEST_INT.  It's just that MAVProxy and MP both 
"auto-upgrade"...

peterbarker, 2019-12-12:
On Wed, 11 Dec 2019, Don Gagne wrote:

> I do wonder though if not using MISSION_ITEM_INT should just be deprecated completely. And all GCS should just use that only and not bother with checking
> capability or whatever?

@WickedShell and I discussed it a little.

Problem is - when do we stop requesting items using MISSION_ITEM and go 
straight to requesting MISSION_ITEM_INT?  Using the heuristic that the GCS 
has previously downloaded from you using MISSION_REQUEST_INT doesn't seem 
a good solution to me - and doesn't get anywhere towards elimination of 
MISSION_ITEM.

Essentially the only change may be a breaking change for 
non-MISSION_ITEM_INT-using GCSs.

hamishwillee, 2019-12-13:
Let me know when you reach a conclusion. FWIW IMO it would be a "good thing" to formally deprecate [MISSION_ITEM](https://mavlink.io/en/messages/common.html#MISSION_ITEM) and just use [MISSION_ITEM_INT](https://mavlink.io/en/messages/common.html#MISSION_ITEM_INT).

DonLakeFlyer, 2019-12-13:
> FWIW IMO it would be a "good thing" to formally deprecate [MISSION_ITEM](https://mavlink.io/en/messages/common.html#MISSION_ITEM) and just use [MISSION_ITEM_INT](https://mavlink.io/en/messages/common.html#MISSION_ITEM_INT).

Next week when I get back home I"m going to put together a proposal on Mavlink to do just that. I see no reason why MISSION_ITEM should be supported any longer.

DonLakeFlyer, 2019-12-24:
I changed QGC to make MP/MavProxy for ArduPilot. I also posted the proposal in mavlink

DonLakeFlyer, 2020-03-23:
So making this change broke compatability with old Solos which are running 3DR firmware. They can't handle getting MISSION_ITEM_INT in all cases. The correct fix I believe is to check the capability bit as well as whether this is ArduPilot firmware. This is likely how it should have been done in QGC in the first case.

If the MISSION_ITEM_INT capability is set and the firmware is ArduPilot then auto-upgrade to MISSION_ITEM_INT on the way up. An old Solo will report lack of support for MISSION_ITEM_INT.

If the capability bit is not set, then don't do that.

@rmackay9, @peterbarker This makes sense doesn't it?

peterbarker, 2020-05-08:
On Mon, 23 Mar 2020, Don Gagne wrote:

> So making this change broke compatability with old Solos which are running 3DR firmware. They can't handle getting MISSION_ITEM_INT in all cases. The
> correct fix I believe is to check the capability [...]

That sounds correct.  'though I'm not sure why you need the "and the 
firmware is ArduPilot" in there - the capability bit ought to be enough?

Peter