mavlink/qgroundcontrol #8087: Mission commands with terrain altitudes are being uploaded as absolute altitudes when uploaded to AP

rmackay9, 2019-12-04:
We've noticed recently that when a user creates a mission command with altitude frame is set to "Terrain" that it's actually being sent as an absolute altitude.

ArduPilot supports mission commands with the altitude specified as an altitude above terrain so we would much prefer if the alt-frame was sent as MAV_FRAME_GLOBAL_TERRAIN_ALT

Here is the corresponding issue on the ArduPilot issues list: https://github.com/ArduPilot/ardupilot/issues/12917

rmackay9, 2019-12-09:
@DonLakeFlyer, any thoughts on this one?  We don't think we can fix it on the AP side because as far as we can tell QGC is not sending the terrain alt.

DonLakeFlyer, 2019-12-10:
QGC does not support MAV_FRAME_GLOBAL_TERRAIN_ALT frame. The "Above Terrain" support is a calculated absolute height based on terrain data and the requested amount above terrain. The docs specify that ArduPilot terrain following is not supported. But it's in release notes as opposed to main docs. I think MP has a sort of similar thing with "Verify Heights" or something like that.

DonLakeFlyer, 2019-12-10:
Reopening since I thought I had specifically put in some warning text when using ArduPilot and Above Terrain to say that this does not support ArduPilot terrain following.

DonLakeFlyer, 2019-12-10:
Hmm, seems to have gone missing in action. Need to look into adding that again.

peterbarker, 2019-12-10:
On Tue, 10 Dec 2019, Don Gagne wrote:

> QGC does not support MAV_FRAME_GLOBAL_TERRAIN_ALT frame. The "Above Terrain" support is a calculated absolute height based on terrain data and the
> requested amount above terrain. The docs specify [...]

Ah, thanks!

So what it comes to is that QGC calculates a terrain offset and sends that 
to the autopilot as an AMSL height.

That's... interesting.  It immediately means you can't round-trip that 
from the autopilot, of course - but, then, your enhanced GCS-side items 
make that difficult anyway.

I imagine, then, that QGC also can't supply terrain tiles to the autopilot 
when requested?

Do you have a fundamental problem with the concept of a "terrain-relative 
altitude", or is just something which has never really popped up to the 
top of a long list?

DonLakeFlyer, 2019-12-10:
> It immediately means you can't round-trip that from the autopilot, of course

That is correct. If you download it back you'll just get an absolute alt.

> I imagine, then, that QGC also can't supply terrain tiles to the autopilot when requested?

Correct.

> Do you have a fundamental problem with the concept of a "terrain-relative altitude", or is just something which has never really popped up to the top of a long list?

Just haven't gotten around to it yet.  I was planning on getting it done in this next stable going out (which I'm trying to wrap up/release in Jan) but ran out of time farting around with trying to change to a different terrain data server mechanism at the same time. It will be on the top of the list for next release.

DonLakeFlyer, 2019-12-11:
Warning is in. So this is working correctly according to current support

closed 2019-12-11 by DonLakeFlyer