mavlink/qgroundcontrol #8096: latest daily builds: UI Improvements needed

dkati, 2019-12-10:
## Expected Behavior

## Current Behavior
1. on the right widget (which shows the commands & waypoints), clicking submenu (camera or vehicle info etc) makes the arrow indicator to disappear. although its there since clicking back , shrinks the submenu

2. on dropdown menus, margins are strange (http://prntscr.com/q4w5lm)

3. mission planner: from right widget: Clicking a waypoint , makes the corresponding dot on the map, green. Although if u click a command (i.e. trigger cam) that is assosiated to a waypoint,the waypoint dot does NOT being green.so u are kinda lost

4. 'L' letter on Launch dot, needs more space. "Increase circle's radius?"

5. About the "new" '+' icon between waypoints: While it works for waypoints, it doesnt work when the Launch is connected to the first waypoint...so u cant waypoint between launch and 1st waypoint

## System Information
When posting bug reports, include the following information
- Operating System: Windows10
**- QGC Dev version: HEAD:916f9938e 2019-11-30 20:35:03 +0000**
- QGC build: daily
- Flight Controller: doesnt matter
- Autopilot (with version): doesnt matter

DonLakeFlyer, 2019-12-10:
> * on the right widget (which shows the commands & waypoints), clicking submenu (camera or vehicle info etc) makes the arrow indicator to disappear. although its there since clicking back , shrinks the submenu

This is on purpose to keep the UI uncluttered with a big pile of little arrowheads.

DonLakeFlyer, 2019-12-10:
> * While it works for waypoints, it doesnt work when the Launch is connected to the first waypoint...so u cant waypoint between launch and 1st waypoint

Launch will only be connected to the first item if the first item is a Takeoff item. And if the first item is a takeoff item you can't put a waypoint before it. This is correct behavior. Is there some other problem?

DonLakeFlyer, 2019-12-21:
These should be fixed now in pulls. There is another item which covers the option to turn on numbering.

closed 2019-12-21 by DonLakeFlyer