mavlink/qgroundcontrol #8105: Pl update mavlink module

olliw42, 2019-12-11:
May I kindly ask to update the mavlink module?

This would help with things like PR https://github.com/mavlink/qgroundcontrol/pull/8007.

THX :)

DonLakeFlyer, 2019-12-11:
Anybody can update to mavlink head. Just do a pull in the submodule dir and submit a pull request.

olliw42, 2019-12-11:
thx for the quick response.

I hear your words but am not sure I really understand. Git submodules and how to handle them properly have remained a mystery to me. Sorry for my limitations. 

I'm worried I rather would create more problems than helping resolve. So, in trying to make sense of your words I did this:

created a branch mavupdate-01
cd'ed to libs/mavlink/include/mavlink/v2.0, showing me (6b8bb7c...)
entering git pull throws an error "You are not currently on a branch", so I did:
git checkout master
git pull 
cd'ed back
commited
pushed branch

which results in PR https://github.com/mavlink/qgroundcontrol/pull/8106. Pl carefully double check if this PR does what it's supposed to do.

DonLakeFlyer, 2019-12-11:
> Git submodules and how to handle them properly have remained a mystery to me.

Nobody understands them. You have to search it each time.

closed 2019-12-11 by DonLakeFlyer

olliw42, 2019-12-11:
you made me laugh now ... :):):) THX

from the fact you merged I at least can conclude that the above recipe does the trick. This was a useful lesson. Thx.