mavlink/qgroundcontrol #8118: Durandal/QGC - Need to add alert message for user running old version of software

jinger26, 2019-12-12:
[Durandal](https://px4.io/new-hardware-holybro-durandal/) is the new flight controller by Holybro. It is supported only with QGC 3.5.6 

But when a user connects the flight controller, he/she might not know that QGC needs to be updated to 3.5.6 for support and be blocked. 

We need an alert message for such user who is running an old version of QGC to update the software to the right version.

From Lorenz: I believe this could be a fairly limited effort, simply pulling a file on startup remotely and checking a version string against the version string in the binary. 
And then that behavior would need to be configurable so people can turn it off.

DonLakeFlyer, 2019-12-16:
> We need an alert message for such user who is running an old version of QGC to update the software to the right version.

Unless someone invents a time machine I don't see how I can go back an update an old version of QGC?

DonLakeFlyer, 2019-12-17:
@LorenzMeier Wait are you just talking about QGC letting people know there is a new version? That is handled by updating https://qgroundcontrol.s3-us-west-2.amazonaws.com/latest/QGC.version.txt

DonLakeFlyer, 2019-12-17:
Which didn't get updated. I'll fix it.

DonLakeFlyer, 2019-12-17:
Updated. I assume that's all you wanted?

closed 2019-12-17 by DonLakeFlyer