mavlink/qgroundcontrol #8129: latest daily: can't connect to autopilot over USB

Antiheavy, 2019-12-16:
I can't connect to an autopilot over USB using the latest daily build on Window's.  Tried on multiple window's machines with multiple autopilots running multiple versions of PX4 firmware.  Everything works fine with an older daily build from 2 weeks ago.

I wonder if it's a USB-serial driver issue with the change to 64-bit?  https://github.com/mavlink/qgroundcontrol/pull/8128

DonLakeFlyer, 2019-12-16:
Ah Windows, how you love to make me crazy!

DonLakeFlyer, 2019-12-16:
Ah, f'ing Qt. It all works fine in 64 bit 5.11.3 but fails with 5.12.5. Good thing 5.12 is LTS, piece of crap!

DonLakeFlyer, 2019-12-16:
https://bugreports.qt.io/browse/QTBUG-78891

DonLakeFlyer, 2019-12-16:
Verified that it works in 5.12.6 which of course is not available in appveyor yet.

DonLakeFlyer, 2019-12-16:
Will need to install 5.12.6 on appveyor manually.

DonLakeFlyer, 2019-12-17:
Should finally be working

closed 2019-12-17 by DonLakeFlyer

Antiheavy, 2019-12-17:
> Should finally be working

yep working in today's daily.  thanks!