mavlink/qgroundcontrol #8137: Video: TCP video hangs QGC

DonLakeFlyer, 2019-12-19:
* Boot QGC
* Set video to TCP
* Set IP to 192.168.143.200:3001
* You get ```VideoReceiver::start() hardware decoding not available``` console output and QGC is now hung

DonLakeFlyer, 2019-12-19:
@antiheavy This directly affects Sentera builds. Prevents me from bringing sentera master up to date and fixing up iOS problems.

DonLakeFlyer, 2019-12-19:
@dogmaphobic @jaxxzer This is a fairly critical problem for me since it's blocking. Not sure where new gstreamer stuff stands as to whether it's soon enough to retry with that or not. If not, I could really use someone to help figure this out.

Antiheavy, 2019-12-20:
> @Antiheavy This directly affects Sentera builds. Prevents me from bringing sentera master up to date and fixing up iOS problems.

maybe temporarily disable for now while a fix is being worked on?

DonLakeFlyer, 2019-12-20:
@Antiheavy Might as well to make forward progress

andrewvoznytsa, 2019-12-21:
@DonLakeFlyer, do you stream from camera or some test gstreamer setup? If gstreamer - can you please share your command line? I'm looking for some way to replicate that issue on my side (and prefer to have exactly the same setup, if possible) and fix.

DonLakeFlyer, 2019-12-21:
Just follow the steps listed above. This is without a vehicle connected or anything actually streaming on TCP.

DonLakeFlyer, 2019-12-21:
@andrewvoznytsa If you can get past that and want to test real TCP then hopefully someone from Sentera can help with a test setup. @Antiheavy?

DonLakeFlyer, 2019-12-21:
I think this is the pull which started to add TCP support way back when: https://github.com/mavlink/qgroundcontrol/pull/5461

andrewvoznytsa, 2019-12-21:
Looks like tcpclientsrc (gstreamer plugin) is connecting to streaming service using in a blocking way. It works if your streaming service is responding and QGC is configured properly.

This does not mean that gstreamer code is perfect - just giving more details. Perhaps it is no so blocking? (still looking for quick and stable solution)

DonLakeFlyer, 2019-12-21:
Seems odd that this used to work? Not sure what changed along the way.

DonLakeFlyer, 2019-12-21:
Just to make sure I just verified that Stable 3.5.6 works fine.

andrewvoznytsa, 2019-12-21:
3.5.6 has workaround - it is checking if server is alive and if it is - connects. I'm checking for easiest way to get it back...

andrewvoznytsa, 2019-12-21:
Sounds like easiest way to get it back is code cherry picking. I'm preparing new PR.

dogmaphobic, 2019-12-21:
The test to see if the "server" is present was also used for RTSP. It looks like it was removed in this PR: https://github.com/mavlink/qgroundcontrol/pull/7435

andrewvoznytsa, 2019-12-21:
Exactly - I'm trying to restore that piece of code without rolling back this PR. Will work as before - for TCP cases (TCP/MPEG TS and RTSP).

Antiheavy, 2019-12-21:
We are definitely available to test any fixes on our hardware.

andrewvoznytsa, 2019-12-21:
Here is PR https://github.com/mavlink/qgroundcontrol/pull/8148. Tested TCP and UDP cases. Did not test RTSP - I don't have RTSP setup right now but I think it should just work.

DonLakeFlyer, 2019-12-22:
Verified Sentera build works now

closed 2019-12-22 by DonLakeFlyer