mavlink/qgroundcontrol #8142: ArduPilot: Implement new large vehicle mag val support

DonLakeFlyer, 2019-12-20:
From Tridge:

ArduPilot master now supports a new type of mag cal which is much
simpler for users of large vehicles. It works by the user pointing the
vehicle in a known direction then asking for a "fixed yaw mag cal"
giving that known yaw.

You can see an implementation of this in MAVProxy here:

 https://github.com/ardupilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_calibration.py#L183

you use "magcal yaw XXX" where XXX is the known yaw.

The calibration requires GPS lock, and uses the WMM tables to work out
the X,Y,Z offsets.

It would be nice to support this in all ArduPilot GCS implementations,
as it is much easier for large vehicles.

dagar, 2020-09-04:
PX4 has support for this as well, but it's so far a bit of a hidden feature for developers bench testing.

``` Console
commander calibrate mag quick
```

I'll open a Mavlink PR to move `MAV_CMD_FIXED_MAG_CAL_YAW` to common.

DonLakeFlyer, 2020-09-04:
Sorry I had lost track of this request. I'll add support for it.