mavlink/qgroundcontrol #8163: Ground Control 2 distance sensor

robomoticchile, 2019-12-23:
## Expected Behavior

I have on my drone a front and down distance sensor, but on the tablet I only can see one (the one on rngfnd1), no matter wich one I sellect.

As you can see in the MP, I can see the front and down sensor, but in the Qground control only shows the down sensor.

http://190.100.113.54:16211/archivos/pictures/QGC1.jpg
http://190.100.113.54:16211/archivos/pictures/QGC2.jpg

The same happends with GQC on windows

## Current Behavior
See as much as distance sensor we have

Here is the Log with the rngfnd 1 and 2 info.

http://190.100.113.54:16211/archivos/logs/rngfnd/

DonLakeFlyer, 2019-12-23:
@robomoticchile I need a gcs telemetry log not a binary vehicle log.

robomoticchile, 2019-12-23:
Im sorry...
Here is it!

[distanceSEnsorsLog.zip](https://github.com/mavlink/qgroundcontrol/files/3995903/distanceSEnsorsLog.zip)

DonLakeFlyer, 2019-12-24:
Thanks for the bug report

closed 2019-12-24 by DonLakeFlyer

robomoticchile, 2019-12-26:
Thansk Don!

Im very sorry to bother you again, but how can aply the fix? I need it to aply to the HereLink QGC.

I appreciate your help, thanks!

DonLakeFlyer, 2019-12-26:
Sorry I'm not involved in HereLink. You'll need to talk with them about that. Not sure where you do that

robomoticchile, 2019-12-26:
thanks, i understand. And if I want to aply to my androdi tablet?

DonLakeFlyer, 2019-12-27:
You can download APKs from here: https://docs.qgroundcontrol.com/en/releases/daily_builds.html