mavlink/qgroundcontrol #8187: MAV_MISSION_INVALID_PARAM4

kristoferahlqvist, 2019-12-27:
## Expected Behavior
It should upload mission

## Current Behavior
When I use camera photo in a mission I get param4 has an invalid value (MAV_MISSION_INVALID_PARAM4). Param4: NaN

Even if I manually set param4 to 0 on all waypoints I get the same error.

## Steps to Reproduce:
1. Make a simple mission with camera photo
2. Try to upload
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: MiuI 11.05
- QGC Version: 3.5.4-1597
- QGC build: daily but I tried stable as well
- Flight Controller: Pixhawk I
- Autopilot (with version): Ardupilot Copter 3.6.8

## Detailed Description
I've built a drone to drop a cargo with a servo that is triggered by take photo

If I use take photo command in mission I get that param4 NaN error even if I manually set param4 to 0.0

## Log Files and Screenshots
[image]

DonLakeFlyer, 2019-12-27:
Should be fixed in daily builds

closed 2019-12-27 by DonLakeFlyer

kristoferahlqvist, 2019-12-27:
> Should be fixed in daily builds

Well it ain't ☹️

DonLakeFlyer, 2019-12-27:
Sorry I missed you listed daily builds as well

DonLakeFlyer, 2019-12-27:
Can you attach a plan file that files?

kristoferahlqvist, 2019-12-27:
[test.plan](https://gofile.io/?c=MMyUxe)

kristoferahlqvist, 2019-12-29:
Tried daily build 1622 but now I another fault.
[image]

DonLakeFlyer, 2019-12-29:
Looks like ArduPilot still doesn't support those commands. The camera commands were added to/updated ages ago (over a year ago if I remember correctly). You should file a report against ArduPilot if you want support for this.

kristoferahlqvist, 2019-12-29:
Ok, weird. Cause it works if I use the "Trigger Camera" button in gqc, but not if I plan it in a mission

DonLakeFlyer, 2019-12-29:
Trigger camera button uses the old mavlink MAV_CMD_DO_DIGICAM_CONTROL camera which I believe ArduPilot does support. You can add these manually into your mission as individual commands. But you won't be able to use the stuff that is intergrated into the waypoint editor.

DonLakeFlyer, 2019-12-29:
I do think the trigger distance item is supported by ArduPilot. Just not take photo, start/stop video and camera mode.

DonLakeFlyer, 2019-12-30:
I had entered an issue on these commands against ArduPilot ages ago. Let me find it and ask them for support again.