mavlink/qgroundcontrol #8190: "Bat2 Voltaje" it is not displayed in the QGC sofware

dronercheliservice, 2019-12-30:
Hi,
Battery_voltage_2 do not show on the QGC scree, only show "-1.00". 
The same happens if I use the HereLink Beta2 GCS.

If I use Mission Planner software, voltage 2 appears correctly "49.60v".

- Operating System: Windows 10 (and also HereLink)
- QGC Version: e.g. 3.5.6
- QGC build: daily (and also stable)
- MP version: 1.3.68
- Flight Controller: Pixhawk 2 CUBE black
- Autopilot software: Arduplane 4.0.1

[image]

[image]

[image]

[image]

See attached telemetry log of my last test flight.

[preflight_perf10_03_38.txt](https://github.com/mavlink/qgroundcontrol/files/4009587/preflight_perf10_03_38.txt)

DonLakeFlyer, 2019-12-30:
@dronercheliservice How come you close this?

> GITHUB does not allow me to upload the datalog files: ***.BIN or ***.PX4LOG

Just pu the file on whatever file sharing service you want to use and link to it. Also I need a telemetry log from QGC not a binary vehicle log.

dronercheliservice, 2020-01-02:
Please see this Logs:

https://drive.google.com/open?id=1BbzCF5od6MGAz7E3ue-g-iaiSCGntvH8

DonLakeFlyer, 2020-01-02:
I need a telemetry log. They are files which QGC saves to the QGroundControl/Telemetry direcotry and ends with an extension of .tlog. Not quite sure what a .px4log file is or where it comes from?

dronercheliservice, 2020-01-05:
Telemetry directory is empty. 
I don´t have any *.tlog file

DonLakeFlyer, 2020-01-07:
Sorry, telemetry logs by default are only saved if the vehicle is armed. Either arm the vehicle, or you can turn that option of from General settings.

dronercheliservice, 2020-01-11:
the vehicle is armed, but the logs does not appear, the directory is empty.
BTWthe directory "Telemetry" only exist at Herelink unit but if i use QGC on Windows the TELEMETRY directory does not exist.

DonLakeFlyer, 2020-01-15:
Do you see any of the QGroundControl directories. They should be under "My Documents" in windows.

hamishwillee, 2020-01-16:
@DonLakeFlyer FMI, when did dual battery support go in, and does it work for both PX4 and ArduPilot? I'm looking at https://docs.qgroundcontrol.com/en/SetupView/Power.html and it doesn't mention this. Does the status icon for battery also display multiple batteries. Is it only 2 batteries supported or does this handle more, if more exist?

DonLakeFlyer, 2020-01-18:
This is just about battery values available to the instrument panel. I added two battery support to that ages ago based on BATTERY_STATUS.id which is likely incorrect. Although there is ArduPilot multi-battery Power setup. This does not exist for PX4. And also toolbar status is not multi-battery.

dronercheliservice, 2020-02-01:
Hi DonLakeFlyer, see link attached with TLOG files:
https://drive.google.com/open?id=1D0en4DMjrz4UX-HEbMrNfZ7KCm5CX7ee

DonLakeFlyer, 2020-02-07:
Perfect thanks. I see the problem. Now I need to figure out how to fix it.

DonLakeFlyer, 2020-02-08:
@dronercheliservice This should be fixed in latest daily. Can you verify?

closed 2020-02-08 by DonLakeFlyer

dronercheliservice, 2020-02-09:
Hi DonLakeFlyer,
now the battery2 value is shown on QGC software using windows 10, many thanks ;)
How can I update my Herelink with this latest version?

DonLakeFlyer, 2020-02-09:
> How can I update my Herelink with this latest version?

Sorry I don't know much of anything about HereLink development. You'll have to talk to them about that.

dronercheliservice, 2020-02-10:
ok, thanks.