mavlink/qgroundcontrol #8192: Multiple missions issue

dogmaphobic, 2019-12-30:
This is a bit convoluted to explain but simple to reproduce.

Start QGC and connect to multiple vehicles:
[image]

Select Vehicle 1, switch to the Plan View, add 2 waypoints and upload the "mission" to the vehicle:
[image]

Back to the Fly View the mission is ready to be started and everything is fine:
[image]

Now select Vehicle 2:
[image]

Switch to Plan View and add a single way point. This should be the first waypoint for Vehicle 2:
[image]

Upload that "single-waypoint-that-isn't-single" mission to Vehicle 2, switch back to Fly View and start the mission. Both vehicles are flying to the Vehicle 1's second waypoint:
[image]

Once Vehicle 1 reaches the second waypoint (its last), it stops as expected and Vehicle 2 proceeds to its "first" waypoint.
[image]

Is this expected? How do I get multiple vehicles to go to different places?

DonLakeFlyer, 2019-12-30:
> Switch to Plan View and add a single way point. This should be the first waypoint for Vehicle 2:

When you upload a Plan QGC always uploads the entire Plan. Hence Vehicle 2 gets the whole Plan. Not just the last waypoint you added. And Vehicle 1 still has what it had before.

Creating plans with multiple vehicles connected is not super easy. The Plan view does not track current vehicle switches. The reason for this is that trying to do that and deal with a possibly dirty plan on the vehicle you are switching away from is a nightmare. And given the fact that multiple vehicle support is used almost never it's not a focus to make easier.

In order to send Plans to multiple vehicles you just need to understand that whatever you are seeing in the Plan view is going to be sent to the current vehicle.

DonLakeFlyer, 2019-12-30:
> Now select Vehicle 2:

There may be a bug here. Assume Vehicle 2 at this point has no mission items on it. Is that correct? Yet a part of the Plan for Vehicle 1 is still being shown.

dogmaphobic, 2019-12-30:
Got it. The way to solve that would be to have an instance of the mission controller on each vehicle. That way it's always in sync with that vehicle. The Plan View would then point to `activeVehicle.missionController` instead.

> There may be a bug here. Assume Vehicle 2 at this point has no mission items on it. Is that correct? Yet a part of the Plan for Vehicle 1 is still being shown.

Correct. This is a first boot for everything. Nothing had any mission previously loaded. The mission you see when vehicle 2 is selected is the one for Vehicle 1. I didn't think of it as a bug. It correctly shows an existing mission. If you were to only see the mission for the current vehicle, we would then need a way to allow multiple vehicle selection so we could see them all.

DonLakeFlyer, 2019-12-30:
> The mission you see when vehicle 2 is selected is the one for Vehicle 1.

Ok, I remember. Originally I had only the active vehicle showing the mission. At some point I changed it. But as it currently stands it is only half-assed implemented. For example rally points and geofence only show for current vehicle. Hard to say what is correct because things will get cluttered really fast if you show everything.

DonLakeFlyer, 2019-12-30:
> The way to solve that would be to have an instance of the mission controller on each vehicle.

Which is how Fly view works, but that's a much simpler case. That's how to make it work for Plan but just the tip of the icerberg for problems.

DonLakeFlyer, 2019-12-30:
So anyway, as it currently stands this is working correctly.

closed 2019-12-30 by DonLakeFlyer