mavlink/qgroundcontrol #8200: Graphing TODO

LorenzMeier, 2020-01-02:
* 32bit disassembly of MAVLink packets: This needs further scrutiny as it could be an issue for Autopilots as well.
  * A plot doesn't auto-scale if you enable it and the initial value doesn't change

dogmaphobic, 2020-01-03:
Regarding 32-bit MAVLink message handling, this is where it fails:

dogmaphobic, 2020-01-03:
Regarding MAVLink message handling on 32-bit ARM. This is where it is failing:
https://github.com/mavlink/qgroundcontrol/blob/master/src/AnalyzeView/MAVLinkInspectorController.cc#L370
```
signal 7 (SIGBUS), code 1 (BUS_ADRALN), fault addr 0xdfbecb2a
```
It seems to happen only with the [DISTANCE_SENSOR message](https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR)

The memory alignment violation happens when accessing `horizontal_fov`.

This is good:
```
    float foo = distanceSensor.horizontal_fov;
    qDebug() << foo;
```
This is bad:
```
    const mavlink_message_info_t* msgInfo = mavlink_get_message_info(&message);
    uint8_t* m = reinterpret_cast<uint8_t*>(&message.payload64[0]);
    foo = *(reinterpret_cast<float*>(m + msgInfo->fields[8].wire_offset));
    qDebug() << foo;
```
```
QString foo;
float* pf = (float*)(void*)&distanceSensor.horizontal_fov;
foo.sprintf("%p", (void*)pf);
qDebug() << foo;
uint8_t* msgPayload = reinterpret_cast<uint8_t*>(&message.payload64[0]);
const mavlink_message_info_t* msgInfo = mavlink_get_message_info(&message);
const unsigned int offset = msgInfo->fields[8].wire_offset;
pf = reinterpret_cast<float*>(&msgPayload[offset]);
foo.sprintf("%p", (void*)pf);
qDebug() << foo;
```
Produce:
```
D QGroundControl: "0xdfe06f42"
D QGroundControl: "0xdfcc7b2a"
```
```
>>> a = 0xdfe06f42
>>> a % 4
2
>>> a = 0xdfcc7b2a
>>> a % 4
2
```
Neigher of which is 32-bit aligned. Both are only 16-bit aligned. Yet, the first is fine and the second causes a fault.

dogmaphobic, 2020-01-03:
> * A plot doesn't auto-scale if you enable it and the initial value doesn't change

It will eventually fix itself when the buffer wraps around.

dagar, 2020-01-03:
To get around the unaligned memory access either memcpy instead of the reinterpret cast or work directly with the struct (which is packed). When you take the address of a packed member the reduced alignment of what you're pointing at is lost.

We've hit various related issues in PX4/Firmware. The warnings `-Wcast-align` and `-Waddress-of-packed-member` can sometimes be helpful, but because everything is packed in mavlink you just need to be careful to copy the data out before doing anything with it.

dogmaphobic, 2020-01-03:
I was afraid moving bytes around was the only option. I’m handling messages in the thousands per second. I need to move that to a worker thread.

LorenzMeier, 2020-01-03:
@dogmaphobic I would test for alignment per field and then only manually unpack if the alignment check fails. That way you preserve efficiency. The alignment test should in theory be compile time (if done correctly), so you should end up with "automatically" getting the same code as now, just in rare instances special handling.

dogmaphobic, 2020-01-03:
It's not that simple, given you don't really know anything about alignment until you receive the message. With that said, I don't really need to process the message unless it is being consumed (either the active message in the MAVLink Inspector or a source for graphing). That should reduce the load considerably and must be done any way.

dogmaphobic, 2020-01-03:
I've dealt with message disassembly here: https://github.com/mavlink/qgroundcontrol/pull/8202
As I mentioned above, avoiding handling the messages when not needed was a much greater (positive) impact than the memcpy penalty.

Closing.

closed 2020-01-03 by dogmaphobic