mavlink/qgroundcontrol #8203: SiK radios and Mavlink2 not working really

olliw42, 2020-01-04:
I do have a copter which uses SiK 433MHz radios, and I can't get it working with QGC. QGC shows a warning that it uses Mavlink1 and also its console logging say that it is switching to Mavlink1. In addition the connection is very fragile and connection gets lost often and repeatedly, and further strange things happen. I don't have any issue when using MissionPlanner with the same setup, it just runs "for ever", and one can prove that Mavlink2 messages get through the SiK radios by the fact that MP receives messages with ID larger than 256, such as e.g MOUNT_ORIENTATION.

I use ArduCopter 4.0, have a STorM32 gimbal which is emitting Mavlink2 messages connected to it, use a QGC daily build of some days ago, and have just updated the SiK radios to the latest firmware with MissionPlanner.

My impression is that the issue is that the logic of QGC for switching to Mavlink1 or Mavlink2 isn't doing the right thing and makes things hang up. In MissionPlanner I can see a vehicle 51 with component ID 68 emitting a heartbeat, which is obviously the SiK radio, and it's maybe this heartbeat which confuses QGC. MissionPlanner's logic seems to make all parties happy and use Mavlink2.

Many thx!
Olli

[image]
[image]
[image]

auturgy, 2020-01-06:
The SiK firmwares from QGC and Missionplanner are different: Missionplanner uses the Ardupilotmega “RADIO” message (id 166), QGC uses the Common “RADIO_STATUS” message (id 109). It would surprise me (as I’d have thought it would have come up before), but as you’re using ArduPilot perhaps try the Missionplanner SiK firmware, and see if the problem persists?

olliw42, 2020-01-06:
oh ... thanks a lot for that info, it certainly is total news to me, THX

I actually tried to use QGC to update the firmware ... however, I have never ever managed to get QGC to flash any firmware for anything on my place (yes, I tried all USB ports I have) ... not pixhawks1, not pixracers, not AUAVX2s, not cubes ... and now I know also not SiKs. I have no idea why this is so, and it certainly doesn't belong to here, just as an explanation for why while your suggestion is great I won't be able to follow it ;)

from a broader perspective I would find it MOST inconvenient to always having to reflash the SiKs when I swap between QGC to MissionPlanner (which I'm kind of doing at ms rate), I wonder if it wouldn't be appropriate to make both GCSes understand both. 

I guess I will give it a chance and see if I can make QGC to be also be happy with RADIO-166 ... :)

thx, sir

auturgy, 2020-01-06:
It may not be the problem: as I said, I would have thought someone else would have seen that problem by now, but something perhaps worth eliminating.

olliw42, 2020-01-06:
someone has to be this "someone else", maybe it just happens to be me :):)
thx anyway, it's a very useful info

auturgy, 2020-01-06:
No idea re Crossfire. Those particular messages are injected by the radios not the autopilot, so it would depend on the TBS firmware.  I’m can’t  recall any radios apart from SiK/RFD that use them.

olliw42, 2020-01-06:
TBS Crossfire is using RADIO_STATUS when in Mavlink mode, that's why I wondered
page 36 in https://www.team-blacksheep.com/tbs-crossfire-manual.pdf

BTW, the MavlinkToPassthrough project also assumes RADIO_STATUS for SiK, but not RADIO
https://github.com/zs6buj/MavlinkToPassthru/blob/master/Mav2PT_v2.48/Mav2PT_v2.48.ino#L473-L482
https://github.com/zs6buj/MavlinkToPassthru/blob/master/Mav2PT_v2.48/Mav2PT_v2.48.ino#L510-L513
https://github.com/zs6buj/MavlinkToPassthru/blob/master/Mav2PT_v2.48/Mav2PT_v2.48.ino#L2435
https://github.com/zs6buj/MavlinkToPassthru/blob/master/Mav2PT_v2.48/Mav2PT_v2.48.ino#L2447
https://github.com/zs6buj/MavlinkToPassthru/blob/master/Mav2PT_v2.48/Mav2PT_v2.48.ino#L2531-L2533

frankly, I think AP could be somewhat less repulsive towards standard MAVLink, but that's certainly OT LOL. 
But it makes me believe that it's probably really something like what you said, which is causing the above issue

olliw42, 2020-01-06:
I just thought I should look into the SiK code ... and found a SiK repo ArduPilot/SiK ... and found in the code

https://github.com/ArduPilot/SiK/blob/master/Firmware/radio/mavlink.c#L43-L45
https://github.com/ArduPilot/SiK/blob/master/Firmware/radio/mavlink.c#L98-L100
https://github.com/ArduPilot/SiK/blob/master/Firmware/radio/mavlink.c#L122

so, this firmware from many years ago at least sends RADIO_STATUS 
 
from the location of that repo I tend to assume that it is the firmware I get when using MissionPlanner 

so: 
> The SiK firmwares from QGC and Missionplanner are different: Missionplanner uses the Ardupilotmega “RADIO” message (id 166), QGC uses the Common “RADIO_STATUS” message (id 109)

is this actually correct?

olliw42, 2020-01-06:
and in the same source I find

`/// we always send as MAVLink1 and let the recipient sort it out.`
https://github.com/ArduPilot/SiK/blob/master/Firmware/radio/mavlink.c#L113

which makes me return to my original suspicion formulated in post #1 :)

DonLakeFlyer, 2020-01-07:
I don't think QGC will care which firmware is running on the radio with respect to mavlink 1/2 transition. The RADIO versus RADIO_STATUS thing is a separate problem in that if you are using radio firmware which used an ArduPilot specific RADIO mavlink message you aren't going to get radio RSSI and telem status reporting with QGC.

To track what is going on with switching from mavlink 1 to 2. Turn on VehicleLog console logging to track what is happening there and then attach the saved console log and I can take a look.

olliw42, 2020-01-11:
sorry for my slow response
I hope these are the logs you were asking for

I had enabled
- VehicleLog
- LinkManagerLog
- MAVLinkLogManagerLog
- MAVLinkProtocolLog

the output with only VehicleLog didn't look very detailed so I thought I should also enable the others. Not sure if that was good or bad.

Here are 3 logs
1. Connection via the SiK up until the connection
[vehiclelog-via-sik-upuntilconnection.txt](https://github.com/mavlink/qgroundcontrol/files/4049728/vehiclelog-via-sik-upuntilconnection.txt)

2. Connection via the SiK for longer times, it catches a "Communication loss" and "Communication regained" which I get repeatedly and relatively frequently with the SiK
[vehiclelog-via-sik-withonelostevent.txt](https://github.com/mavlink/qgroundcontrol/files/4049730/vehiclelog-via-sik-withonelostevent.txt)

3. For comparison a log where I connected via the USB of the pixhawk, just in case
[vehiclelog-via-usb.txt](https://github.com/mavlink/qgroundcontrol/files/4049732/vehiclelog-via-usb.txt)

with either MissionPlanner or with connecting per USB I can run the system indefinitely without issues.

THANKS a lot, Olli

olliw42, 2020-01-11:
in order to test "my theory" that it's the SiK's RADIO_STATUS message sent as MAVLink V1 which confuses QGC, I simply un-commented the line 

https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/MAVLinkProtocol.cc#L327

in order to prevent QGC to switch back to V1, et voila, it works and works and works :)

this furthers the notion that the mavlink version negotation procedures in QGC are somewhat at odds. I remember there had been a similar issue several months back and the solution taken then wasn't satisfying from my perspective. No complain, just to hint that this might be a point which might need some reconsideration.

as regards the issue itself, based on the observed behavior and based on the comment 

/// we always send as MAVLink1 and let the recipient sort it out.
https://github.com/ArduPilot/SiK/blob/master/Firmware/radio/mavlink.c#L113

in the Sik source code, it seems to me that the SiK radios BY DESIGN do NOT switch to sending the RADIO_STATUS as MAVLink V2, and that therefore 
1. the implemented logic for switching V1/V2 necessarily needs to fail
2. a warning "Detected radio still using MAVLink v1.0 on a link with MAVLink v2.0 enabled. Please upgrade the radio firmware" is intrinsically incorrect since no upgrade will ever make it sending MAVLink v2 RADIO_STATUS messages

It looks to me that the radio_status test code thing should just go.

DonLakeFlyer, 2020-01-15:
> in order to test "my theory" that it's the SiK's RADIO_STATUS message sent as MAVLink V1 which confuses QGC, I simply un-commented the line

Thanks for digging into this @olliw42. I still hadn't had time to look a this. I hadn't seen that code in mavlinkprotocol before and it is incorrect. I had fixed up all the other V1/V2 negotiation code but didn't realize what you pointed to existed. Fixing is going to be more involved than what I can do from the road since all the V2/V2 stuff is a nightmare of fragility. So it's will have to wait till I get back and fix it before Stable 4.0 goes out.

olliw42, 2020-01-16:
thx for going to have an eye on it

> Fixing is going to be more involved than what I can do from the road since all the V2/V2 stuff is a nightmare of fragility

yes, I fully understand this, the second half is unfortunately very true

the v1/v2 negotiation "rules" are currently not fully clear and based on too many assumptions of the behavior of the other parts (should be actually a topic for the MAVLink folks)

I believe that you will find that one can't base any decisions on RADIO_STATUS. For the SiKs they are v1 but the SiK happily passes on v2 messages and the current logic fails. For crossfire it is also v1, but crossfire doesn't switch to v2 so here the warning and handling would be correct. But one likely also finds systems which behave as assumed in the current logic. So, all is possible and thus no decision can be based on it. 
I believe that you also will find that one never should step back to v1 if there ever had been a v2 message from a link.

For the moment, once the correct rules for compiling latest QGC are out, I will be happy enough with using my own compile, but the situation is certainly very unfortunate and I thus much appreciate that you have put it to the milestones for the next stable release.

thx

DonLakeFlyer, 2020-01-18:
The v1/v2 negotiation protocol is in the mavlink spec somewhere if you look for it. I’ll have to look at it again to see if it is incorrect with respect to the problems found. I don’t remember anything in there about radio messages affecting things. That could just be a big in QGC as opposed to a spec thing.

olliw42, 2020-01-18:
I know where the v1/v2 spec is :) And I actually also have read it LOL
it doesn't provide sufficient guidance however to establish a generally working setup consisting not only of a autopilot but also other components, IMHO
but that's a topic by its own and largely unrelated to the present issue
the problem here is that the RADIO_STATUS version affects the version of the link

DonLakeFlyer, 2020-02-08:
@olliw42 Can you capture a telemetry log of the case you have where it fails so I can play it back and debug?

DonLakeFlyer, 2020-02-09:
@olliw42 New dailies should be ready in couple hours. Can you verify the change?

olliw42, 2020-02-10:
I'm sorry that I am a bit silent on this, QGC however fell a bit down on my list of things. 

I think I did post some logs in the above.

For said reason, sorry, I'm currently not in a good position to test and verify the change. However, if it's just deactivating it, what it seems to be, then it should work for me since this is EXACTLY what I did to get it working.

DonLakeFlyer, 2020-02-13:
This should be working now

closed 2020-02-13 by DonLakeFlyer